w3c-cg/web-nfc #16: Does the visibilityState of the page have effect on the API behaviour?

anssiko, 2015-04-14:
How should the API behave if the [visibilityState](http://www.w3.org/TR/page-visibility/#dom-document-visibilitystate) of the page is hidden? This is currently unspecified, i.e. considered an implementation detail.

Some existing APIs that make use of hardware devices such as [vibrate()](http://www.w3.org/TR/vibration/#dfn-processing-vibration-patterns) behave differently depending on the visibilityState of the page in order to make it obvious to the user which page is using the device.

The [Security and Privacy doc](http://w3c.github.io/web-nfc/security-privacy.html) seems to suggest the API should be functional only if the page is visible to the user:

> the user needs to be aware of exactly what a web page is intending to do with NFC at any given moment

kenchris, 2015-04-14:
I agree with that. It should only work when visible... then again, we might want to rethink that for card emulation when we support that, so that people can use it to open doors etc without having to make sure the phone screen is on, app is focused etc

zolkis, 2015-04-14:
Some of the functions are indeed affected, like write(), and set/clearPushMessage().
But read notifications for watchers should work even when the page is not running, once the listener/watcher origin was registered. Then, for watch() and clearWatch() the page does not need to be visible, AFAICS.

zolkis, 2015-04-14:
For simplicity we can start with saying that all methods require visibility. The reason is that the above methods require activity from the user both for reading and writing.

anssiko, 2015-04-14:
@zolkis That sounds like a reasonable approach. It'd allow us to enable certain methods for hidden pages later if/when we figure out how to do so without compromising users' security and privacy.

zolkis, 2015-09-23:
Fixed by #50.

closed 2015-09-24 by zolkis