w3c-cg/web-nfc #17: service workers, protocol handlers

zolkis, 2015-04-28:
Are service workers supported, or only visible web pages can handle NFC?

Web NFC implementations could have policies like:
- if there is an open visible page that handles NFC (i.e. has requested an adapter), let it handle it
- otherwise, if there is a single service worker running, let it handle it
- otherwise, if there are multiple service workers, ... <invent a policy>.

Certainly, at any point of time there should be max one page that can handle NFC (reads and writes alike).

Should we use protocol handlers? like: 
```navigator.registerProtocolHandler('web+nfc', 'http://myorigin.com?q=blabla', 'MyApp');```

flaki, 2015-04-28:
Service Workers are event-based background workers, which means most of the time there shouldn't be any service workers running unless they are servicing a request (be it for a page/navigation, or other background tasks like incoming notifications for push messages).

That said, once spun up, service workers remain running for some time solely for optimization purposes (there are ongoing talks for how long they should be kept alive by default, though).

There are two ways to overcome this: one is to have the events baked in to the platform and make service workers capable of receiving incoming NFC data (that would need working with SW spec authors I guess).

The other method is to place NFC comms into their own Web Worker. There is ongoing discussion to make Service Workers to be able to have their own workers/workers shared with pages they control. In this scenario, the NFC comms would be running in a worker thread and only interacting with the page or service workers through this one indirection.

jyasskin, 2015-04-28:
The security model so far requires that the user see a web page while that web page is allowed to access the NFC radio. A service worker is, by definition, not visible, which implies that service workers can't access NFC.

zolkis, 2015-04-28:
Which is also the model used by other NFC API's. The question is whether should we relax that when policies could be defined such that they could guarantee that at most one client can access NFC during a given touch "transaction".

jyasskin, 2015-04-28:
The security model isn't based around "at most one" though: it's based around the user seeing the actor on their screen.

It's definitely worth designing the API so it extends naturally to service workers when we find a way to relax the security model, but I think it's premature to actually relax it yet.

zolkis, 2015-04-28:
I assumed the outcome of applying policies would be to start the most relevant page that will handle NFC. I agree that this would be a v2 feature, I just wanted to make sure the API we are going to propose for v1 is forward compatible with this.

kenchris, 2018-11-30:
Can I close @zolkis ?

closed 2018-11-30 by zolkis