w3c-cg/web-nfc #24: Will there be a way to read the NFCTag's UID?

zolkis, 2015-06-04:
_From @kontza on June 1, 2015 12:27_

I browsed through the specification, and didn't find a mention on how an application could read/acquire the NFC tag's UID. Why is that? My company has deployed tens of thousands of NFC tags to our customers, and for us the most important thing about NFC tags is the UID.

_Copied from original issue: w3c/nfc#91_

zolkis, 2015-06-04:
Do you mean the NDEF ID field? If yes, then - by the current thinking - it's used by the Web NFC implementations, contains the writing scope, and it is visible to webpages. In other words, it's a special use of the ID. The main use case with this API is that web pages should see mainly tags designed for the web browser usage. Reading 'legacy' tags is possible, but subject to security policies. 

However, technically it would be possible to expose the ID field for apps use, if you have a strong use case for it and can solve the security implications.

zolkis, 2015-06-04:
BTW, did you mean to ask on https://github.com/w3c/web-nfc/issues? This working group is closed, and the spec is developed in the https://www.w3.org/community/web-nfc/.
I can migrate this issue there.

kontza, 2015-06-11:
No, I did not mean NDEF-field, I meant the low-level RFID UID. You can see a similar discussion on it on MSDN: https://social.msdn.microsoft.com/Forums/windowsapps/en-US/89c1d752-ea6e-4aaa-bc18-fe162b0b85d8/lowlevel-rfid-reading-in-windows-phone-8?forum=wpdevelop
where user SgtRiggs asks for similar functionality for Windows Phone 8. I banged my head against the wall with WP8 NFC tag APIs on that very same issue: no access to the tag's low-level UID.

Here's an excerpt from the MSDN discussion describing, IMHO, a valid reason to have access to the low-level UID:
> Without the tag's UID or ATR or chip serial number there is no way to detect a clone of the tag versus the original.
> This is important. The class interface is interesting but DOA for serious NFC use without the ability to confirm the authenticity of the tag the phone is reading or publishing.

anssiko, 2015-06-11:
[I recommend responding via GitHub comments using the link below these mails. GitHub comments made on the w3c/web-nfc repo are automatically reflected to the group's public-web-nfc@w3.org mailing list, but replies to these mails are not reflected back to GitHub automatically.]

@adrianhopebailie on [public-web-nfc@w3.org](https://lists.w3.org/Archives/Public/public-web-nfc/2015Jun/0004.html):

>Most NFC use cases assume that the issuer of the tag is not the same entity as the reader of the tag which I suspect is why this has not come up before.

>This is a valid and important use case for tag issuers to be able to verify which physical tag was used.

>From that MSDN issue it sounds like access to the UID is not even possible at the OS level on Windows so not sure how the browser will be able to expose it unless that changes.

zolkis, 2015-06-11:
Thanks Juha for the clarification. I need to understand why this use case is relevant for browsers. Tag issuers may verify physical tags in many other means than via a browser. We need to discuss this use case from security perspective, and also feasibility on various platforms. As a side note, low level access was not among the main objectives of this API.

zolkis, 2015-08-13:
It seems we cannot implement this feature as there appears to be no support for it in native NFC API's.
If ever it will be supported, it will be a function on the NFCAdapter object, returning a Promise which resolves to a DOMString value, or rejects with an exception (NotSupported, etc).

anssiko, 2015-08-14:
I suggest we close this issue since it appears the platform APIs do not expose tag's low-level UID. We can revisit the issue if the situation changes with the platform APIs and we unearth a valuable use case that requires access to such low level information.

beruic, 2017-09-27:
Does anyone know if this has changed in platform API? I have a similar use case as @kontza, where we are trying to make our web site the app for our product as an installable web app.
We currently have a Cordova app solution where the UID is readable, but it would be much preffered to get rid of this.

ClintonGallagher, 2018-05-02:
NFC UID is made of pyrite, aka fool's gold. 

* The syntactical and grammatical format of the UID is not consistent across manufacturers
* The value is not exposed to the browser
* There is no mechanism to make the UID part of the URI query string
* Reading UID requires a proprietary app

Conclusion? NFC UID is pragmatically useless

leolux, 2018-05-02:
@ClintonGallagher
The last three points regarding the UID is only the default behaviour. NXP has a built-in machanism for adding the UID automatically to the URL content of the tag while reading out the URL. It depends on how  your configure your NFC tags to work.

zolkis, 2019-05-24:
Also discussed in #224, as with the v2 API the tag serial number/UID might be exposed when reading a message (if available from the underlying platform).
In general, so far in this API we have exposed NDEF messages, without tag tech details. 
The question is how to expose low level information to web pages. 
IMHO we should keep them separate from NDEF messages.
(work in progress)

kenchris, 2019-08-05:
We support this now

closed 2019-08-05 by kenchris