w3c-cg/web-nfc #26: URL objects

annevk, 2015-07-16:
Please don't use URL objects in data structures: https://url.spec.whatwg.org/#url-apis-elsewhere

(I would also suggest not referencing all of these out-of-date forks of WHATWG Standards. That doesn't seem helpful to readers.)

zolkis, 2015-08-11:
The problem we have in the NFC spec is that we expose incoming data types as 
```
typedef (DOMString or URL or Blob or JSON) NFCData;
```
Now if ```URL``` becomes ```USVString``` then in JS it will be normal string, indistinguishable from text type. That means if we get rid of ```URL``` objects, we need to carry the type of the original information as a property (like ```type: "url", value: "www.mydomain.com"``` vs ```type:"text", value: "www.mydomain.com"```), which is not very web'ish either, then this will roll on and change the whole API. <*scratching head*>

annevk, 2015-08-11:
That looks like a problematic structure either way. You cannot distinguish JSON from a string.

zolkis, 2015-08-11:
In the current implementation, JSON is exposed as Object (property bag) to clients, URL as a URL object, Blob as Object. When a client gets an Object, it can use ```instanceof``` to tell it was Blob; if it was not a blob, then it was JSON. This works now, but I don't know which evil is worse: using URL object, or carrying type along in a property. Any best practice on this nowadays?

annevk, 2015-08-11:
What's the underlying representation you're trying to model?

kenchris, 2015-08-11:
NFC allows you to store different kinds of data. For write, we could just have multiple functions, but I am not sure what would be the best model for read. Push API seems to have http://www.w3.org/TR/push-api/#idl-def-PushMessageData which has different getters.

annevk, 2015-08-11:
Right, what Push does is based on Fetch. Still, what types of data do we have and how are they exposed underneath?

kenchris, 2015-08-11:
It can store values like URL (slightly "compressed" by NFC like the new eddystone URLs from physical web), string, and binary data (mime type). We also use a special record for storing json.

http://w3c.github.io/web-nfc/#web-nfc-payload

annevk, 2015-08-11:
It seems push does not have metadata to include the type. For `Request` and `Response` objects in Fetch that is expected to be delivered through the `Content-Type` header and necessary for the correct extraction of certain types. If you want something analogous I would expect:

```webidl
interface NFCData {
  readonly attribute NFCDataType type;
  ArrayBuffer arrayBuffer();
  ...
  USVString url();
}
enum NFCDataType { "url", "string", "opaque" };
```

domenic, 2015-08-11:
+1 to @annevk's design. Alternately, a design like `{ type: "url", data: ... }` `{ type: "text", data: ... }` seems fine. Anne's design is better to communicate that NFC's metadata is not relevant and you want the ability to deserialize in multiple ways as appropriate for the use case. The `{ type, data }` seems a bit better to give a more direct mapping to NFC (although in that case just exposing TNF and MIME type would be more appropriate).

kenchris, 2015-08-11:
What @annevk suggests is pretty aligned when what I was suggesting originally :-) so I am on board with that. @zolkis what is your take?

domenic, 2015-08-11:
Also I think the proper name for Blob-with-mimetype is File?

annevk, 2015-08-11:
No, Blob-with-filename is File. Blob always has a MIME type (though it may be unknown).

domenic, 2015-08-11:
Right, sorry, I was doing so well for early-morning and then I had to go and blow it :P. I'll go get breakfast and coffee now...

zolkis, 2015-08-11:
@domenic 
> The { type, data } seems a bit better to give a more direct mapping to NFC (although in that case just exposing TNF and MIME type would be more appropriate).

There is some overlap in NFC regarding TNF and MIME types vs mappable JS types, so for *Web* NFC we have already done a slight mapping [here](http://w3c.github.io/web-nfc/index.html#web-nfc-payload). Then, we didn't mean exposing the low level NFC details in this spec - for that we still have the older WG spec which follows low level details very well. This spec is more like "rethinking NFC usage for the Web", and the argument would favor @annevk 's design (indeed we have iterated through a similar design with @kenchris earlier). I'd rather start with
```
enum NFCDataType { "url", "string", "json", "blob" };
interface NFCData {
    readonly attribute NFCDataType type;
    Blob blob();
    JSON json();  // same as 'Object json();'
    USVString url();
    DOMString text();
    // ArrayBuffer arrayBuffer(); // skip this if we don't expose the low level content
}
```
since in that case we have some indication how the content should be handled by the client. Functions may throw type exceptions. But I might miss the details behind the ```"opaque"``` type idea.

Thank you all for helping in this.

domenic, 2015-08-11:
That design sounds pretty good. Although I would include arrayBuffer() if you are including blob(). Otherwise you are just telling users that you hate them and want to subject them to the pain that is FileReader :P.

zolkis, 2015-08-11:
Indeed, even the examples used FileReader :). If we skip the idea of exposing low level content, then we can dedicate arrayBuffer() to Blob(). Or, if we ever wanted to expose that, we could choose a special name for that purpose.

kenchris, 2015-08-11:
So I guess @annevk was suggesting using "opaque", which could then be read with blob() or  arrayBuffer() ?

annevk, 2015-08-11:
Yeah, that was my idea. I'm not sure I'd expose `Blob` if you can do `ArrayBuffer`. Doesn't make much sense to offer the asynchronous variant if you offer the bytes synchronously already. If it's about the `type` that a `Blob` can expose you really ought to expose that somehow. Does NFC carry a MIME type?

kenchris, 2015-08-11:
Yes, it does for binary data

annevk, 2015-08-11:
So was your idea to expose that through `Blob`? It seems exposing it directly on `NFCData` would be simpler?

kenchris, 2015-08-11:
Yeah that is an option. Just 'DOMString mimeType()' ?

annevk, 2015-08-11:
What is the MIME type if the underlying data is a URL or a string? Just blank? It seems you could just offer `type` and either it's the blessed "url" or "string", and a full blown MIME type otherwise. Also, is the underlying MIME type restricted to 0x00 to 0x7F? In that case you want `ByteString`.

kenchris, 2015-08-11:
Are there other existing examples of APIs like this in the web platform?

annevk, 2015-08-11:
`DataTransfer` has something similar with magic "text" (better than string actually) and "url" values that map to text/plain and text/uri-list.

annevk, 2015-08-11:
You could either do that or keep them distinct, with a `type` property getter and a `mimeType` property getter. If the former returns "opaque", the latter will not be the empty string. Kind of depends on how you expect developers will want to branch their code. Seemed reasonable to me that you'd just have:

```js
if(data.type === "url") {
  ...
} else if(data.type === "text") {
  ...
} else if(data.type === "image/png") {
  ...
} else {
  reportError()
}```

zolkis, 2015-08-11:
Since JSON, text and URL could be mapped to MIME types (```application/json```, ```text/plain```, and ```text/url``` indeed we could go with
```webidl
interface NFCData {
    readonly attribute DOMString mimeType;
    ArrayBuffer arrayBuffer();
    JSON json();  // same as 'Object json();'
    USVString url();
    DOMString text();
}
```

kenchris, 2015-08-11:
JSON and text can be mapped to different mime types, I like what @annevk suggested more

annevk, 2015-08-11:
Actually, `JSON` needs to be `any` and it seems I did that wrong in Fetch. `true` or `1` is valid JSON.

zolkis, 2015-08-11:
>JSON and text can be mapped to different mime types

But it's a bit awkward to have 2 type specifiers, like "type" and "mimeType", and I think we could manage with one - this still permits having different MIME types for text and JSON.

kenchris, 2015-08-11:
That was @annevk suggestion.

ByteString type; // "text", "json", "url", or mime type

annevk, 2015-08-11:
Does the NFC protocol have JSON as a native type? Or is that just opaque that happens to utf-8 decode and parse as JSON?

zolkis, 2015-08-11:
Yes, but "text" and "json" have MIME types, and user agents could expose then as such.

kenchris, 2015-08-11:
@annevk the latter. We use a special record for json

domenic, 2015-08-11:
I am a bit confused. The method-based design makes sense only if the same data can be deserialized multiple ways. E.g. I could take the same NFCData and get it as an arrayBuffer, as json, as text, and maybe even as a URL. Is that the case? Or is the design instead that all but one of those methods would throw?

kenchris, 2015-08-11:
@domenic that is the reason why I was a bit ambivalent about this API when I suggested it long ago, but it is consistent with Push API

annevk, 2015-08-11:
@kenchris so why not map directly what NFC does? "text", "url", and everything else is a MIME type plus bytes. Then you have convenience methods that extract the data, some of which will may fail if the data is incorrect.

domenic, 2015-08-11:
Wait, but what is the answer?

zolkis, 2015-08-11:
JSON is in fact Media type as defined in [RFC2046] (TNF=2) with associated MIME type "application/json".

zolkis, 2015-08-11:
The spec should state the conditions of throwing in readers. With the strictest policy, yes, only one would not throw.

annevk, 2015-08-11:
@zolkis there's also e.g., `application/vnd.github+json` and various others JSON MIME types.

domenic, 2015-08-11:
If the design being discussed is that all but one of those methods could throw, then this is a terrible design. Symmetry with Push is not a good reason.

In that case I would suggest instead

```
interface NFCData {
    readonly attribute DOMString mimeType; // or type? or both?
    any data; // ArrayBuffer or USVString or DOMString or JSON
}
```

zolkis, 2015-08-11:
I love this.

zolkis, 2015-08-11:
> there's also e.g., application/vnd.github+json and various others JSON MIME types.

That would be conveyed as Blob with the given MIME type, exposed as
```javascript
{
   type: "application/vnd.github+json",
   data: 0
}
```
and on NFC level it would be a TNF 2 media message with the given MIME type.

kenchris, 2015-08-11:
@domenic How would you deal with the writer functions? Have separate ones for the types? or automatically try to detect what any represents?

zolkis, 2015-08-11:
I think we could do a default mapping based on ```instanceof```.

kenchris, 2015-08-11:
@zolkis i'm just wondering whether it will be too much magic and result in inconsistencies across implementations

domenic, 2015-08-11:
I can't find "writer functions" in the spec but I would expect a symmetric interface: `write({ mimeType: "...", data: "..." })`. No type inference or magic needed. You would define a dictionary counterpart to NFCData for this purpose.

zolkis, 2015-08-11:
Right now we have
```webidl
[NoInterfaceObject]
interface NFCMessage {
    readonly    attribute USVString     scope;
    readonly    attribute sequence<any> data;
    // (DOMString or URL or Blob or JSON)
};
```
and this needs some update now, but the types are perfectly determinable for writing.
If we want to be strict, we can do what @domenic suggests.

zolkis, 2015-08-11:
That is, 
```webidl
interface NFCMessage {
    readonly    attribute USVString     scope;
    readonly    attribute sequence<NFCRecord> data;
};
interface NFCRecord {
    readonly    attribute DOMString    mimeType;
    readonly    attribute any data;
};
```

```

kenchris, 2015-08-11:
How does all this sound to you @annevk  ?

annevk, 2015-08-11:
I would do s/`mimeType`/`type`/. Otherwise this seems fine. `scope` looks suspect too, but I suppose we cannot get rid of that due to service workers?

kenchris, 2015-08-11:
OK, what is the best type for storing the mime type? ByteString as suggested earlier?

annevk, 2015-08-11:
@kenchris again, what does the underlying protocol store it as? If it's bytes, and not utf-8 bytes, `ByteString` would make sense. (We should be as low-level as possible, while within security constraints, and still providing a somewhat attractive API.)

kenchris, 2015-08-11:
Sorry for all the questions :-) I just want to get this right.

'scope' is not for SW, but a way to know which site actually wrote to the TAG, and it allows for the site to use the path to differentiate/id the data they stored. Would be good if you could read the sections about this in the spec

zolkis, 2015-08-11:
> I would do s/mimeType/type/. 

Perfectly fine!

>  scope looks suspect too, but I suppose we cannot get rid of that due to service workers?

Scope needs a serious discussion. See also #22 , and we just closed #15. Hence, Occam would remove it for the moment...

kenchris, 2015-08-11:
@annevk I guess that doesn't just depend on how it is stored on the NFC tags but also how the native NFC APIs expose it. We would have to look at that.

zolkis, 2015-08-11:
As for (MIME) ```type```, IMHO developer convenience should be the priority. It's not a low level information we should try to maintain at low level, but a helper to access the low level information.

kenchris, 2015-08-11:
@annevk on Android, the API exposes it as String (UTF-16):

```
String	toMimeType()
Map this record to a MIME type, or return null if it cannot be mapped.
```

annevk, 2015-08-11:
@kenchris well, `ByteString` also returns a string (for convenience). It's just that the underlying representation is bytes. I guess you're saying that on Android the underlying representation is a string, though is this the C or Java API?

@zolkis it's really more a question about the value range than convenience with ByteString vs DOMString vs UVSString.

kenchris, 2015-08-11:
@annevk the Java API: http://developer.android.com/reference/android/nfc/NdefRecord.html - I haven't looked at whether the NDK exposes NFC or not, and how. @zolkis What does neard expose?

[46 more replies]

closed 2015-09-23 by zolkis