w3c-cg/web-nfc #27: NFC is an abbreviation, don't spell it "Nfc"

annevk, 2015-07-16:
NFC for interfaces and `nfc`/`otherTermNFC` for attributes would be much more consistent with naming elsewhere.

kenchris, 2015-07-16:
Thanks for the report. The platform is pretty inconsistent regarding this. Do we have any guidelines anywhere? Or should abbreviations from now on always be written with uppercase (when not first word)?

annevk, 2015-07-16:
I think only `XMLHttpRequest` got this really wrong.

I don't think we have guidelines, but what I said is what we usually do, @domenic?

domenic, 2015-07-16:
Yep, https://github.com/w3ctag/design-principles/issues/15 is noting that it's uppercase everywhere (except in initial camel case, `nfcStuff`).

kenchris, 2015-07-16:
getElementById. So ID or Id? I think WebBluetooth was using Uuid as well (we had a bug about it). I also think that there are cases of Url. I remember that we looked at it and found many inconsistent cases

domenic, 2015-07-16:
Id is an abbreviation, not an initialism. NFC is an initialism.

If web bluetooth is using Uuid then we should fix that. Url is incorrect if it ever appears; see https://url.spec.whatwg.org/#url-apis-elsewhere.

kenchris, 2015-07-16:
Thanks @domenic we will follow that advise (after vacations end :))

kenchris, 2015-07-16:
So you are saying that it should be ...Id but ...UUID? Then why not UUId?

It would be nice to have examples like these for the guidelines

domenic, 2015-07-16:
Yep, we'll add them to the guidelines.

jyasskin, 2015-07-16:
We fixed Bluetooth in https://github.com/WebBluetoothCG/web-bluetooth/pull/106. Do file bugs if you find other examples.

zolkis, 2015-08-10:
In most cases using the full caps for initialisms is OK, unless adjacent to a next word, when it's both ugly and less readable (consider a hypothetic XMLHTTPRequest). I would argue that though initialisms are all uppercase in free text (NFC, XML, HTTP, UUID etc), but in code (variable and function naming) it would be fine to use CamelCase: for instance, XmlHttpRequest looks better IMHO than either XMLHttpRequest, or XMLHTTPRequest). But tradition is likely too strong for some of the names (even I would blink on seeing DomElement for instance :).

But instead of 'ID' I would use 'Id' for sure.

kenchris, 2015-08-11:
@zolkis we don't really have any problems like that with NFC. I think it is fine to follow what is commonly used, thus all uppercase.

zolkis, 2015-08-17:
Fixed by [this](https://github.com/w3c/web-nfc/commit/1aa321be48cedc0744d3302018c270bdb7cbb195) commit.

closed 2015-08-17 by zolkis