w3c-cg/web-nfc #33: Define "sub-domain match"

jyasskin, 2015-08-17:
Ideally just by reference to an existing algorithm in another web standard.

Alternatively, consider just requiring that the [origins](https://html.spec.whatwg.org/multipage/browsers.html#origin) be equal. The web is moving away from use of sub-domains for security.

annevk, 2015-08-18:
Yeah, if we're going to do eTLDs here that would need to be justified somehow.

zolkis, 2015-08-18:
As a default, we could write the origin of the ~~page~~ browsing context using the API, for instance ```https://mydomain/mypage```.
To make it a bit more useful, we said the page could write any sub-domain match of its own origin, to differentiate between different writes, for instance ```https://mydomain/mypage/mygame1```.
This is more generic than the algorithm for same origin.

annevk, 2015-08-18:
I think you need to read up on "subdomain", "origin", and "path". (Also, a browsing context doesn't have an origin, its active document does.)

kenchris, 2015-08-18:
@zolkis don't you mean 'more specialized path'? A subdomain would be something like hangouts.google.com (a subdomain of google.com)

zolkis, 2015-08-18:
The example above was screwed, but IMO both would qualify for the NFC use cases I described. It is up to us to decide.

kenchris, 2015-08-18:
Can't we say that the url needs to have a path relative to the base url? (or would that allow ../other path)

jyasskin, 2015-08-18:
Don't implement security restrictions based on paths. The origin is the fundamental concept of web security, so that's what your restrictions should be based on.

Anne's point about using the [incumbent settings object](https://html.spec.whatwg.org/multipage/webappapis.html#incumbent-settings-object) is good. It has an [origin](https://html.spec.whatwg.org/multipage/webappapis.html#definitions-2:origin-2).

kenchris, 2015-08-18:
but what about services having different projects under the same domain?

I know github uses subdomains for that, so maybe it is not a problem

jyasskin, 2015-08-18:
They're not secure against each other. For a very few things ([maxScopeString in ServiceWorkers](https://slightlyoff.github.io/ServiceWorker/spec/service_worker/index.html#update-algorithm)) there's some attempt to let servers segregate paths, but that's only when the impact from cross-path attacks is big (persistently overwriting the other path's contents, for SWs). In NFC, the impact is small (reading another path's data), and could also be accomplished with any of the other storage APIs, so it's not worth worrying about.

zolkis, 2015-09-23:
This became obsolete.

closed 2015-09-23 by zolkis