w3c-cg/web-nfc #35: Propose default value for timeout

alexshalamov, 2015-08-18:
At the moment steps for send() method https://w3c.github.io/web-nfc/#the-send-method define that promise should be rejected if peer / tag is not tapped within platform specific timeout (step 10, 11 and 12).

It would be good if spec would propose default value for the timeout, so that it would be consistent between browsers.

kenchris, 2015-08-18:
@annevk what is normal to do in such a situation? default values are preferred?

annevk, 2015-08-18:
For networking timeouts are not defined as far as I know.

jyasskin, 2015-08-18:
[allowed to show a popup](https://html.spec.whatwg.org/multipage/browsers.html#allowed-to-show-a-popup) just says
> The task in which the algorithm is running was queued by an algorithm that was allowed to show a popup, and the chain of such algorithms started within a user-agent defined timeframe.

It gives some example times, but doesn't suggest a particular length for UAs to use.

In Bluetooth, we have a timeout hidden in "perform the General Discovery Procedure", which in the BT spec includes "should continue for a minimum of T<sub>GAP</sub>(gen_disc_scan_min).", which defined as 10.24s. Does the NFC spec have any default timeouts?

zolkis, 2015-09-23:
See also #49. It was chosen not to select a default timeout.
Developers are as such encouraged to specify one.
Underlying platforms and implementations may have default timeouts, overridden by the specified timeout.
If the specified timeout cannot be met, the UA rejects.
Please reopen if this is not good enough resolution.

closed 2015-09-23 by zolkis