w3c-cg/web-nfc #36: website is not a thing

annevk, 2015-08-18:
Please make sure everything you say is established in first principles.

kenchris, 2015-08-18:
What is the right term? web page?

annevk, 2015-08-18:
I think often you mean to say browsing context (or address some part of that concept).

annevk, 2015-08-18:
Note that the same goes for "web app". E.g., "web app's document URL" is pure fiction. Where is it defined that a "web app" has a document URL? And what does that mean?

zolkis, 2015-08-18:
Yeah, these are more colloquial than standardized/canonical names, even though are defined terms. Would be good to use one terminology - suggestions?

annevk, 2015-08-18:
Well, you basically want to look at what these map to in an implementation and where the concepts are defined that this implementation already has. Typically that means studying the HTML Standard.

kenchris, 2015-08-18:
Seems pretty fitting:

http://www.w3.org/TR/html5/browsers.html#windows

A browsing context is an environment in which Document objects are presented to the user. Note: A tab or window in a Web browser typically contains a browsing context, as does an iframe or frames in a frameset.

jyasskin, 2015-08-18:
Here's a first cut at replacing uses of websites in the NFC spec: 

1. The uses in http://w3c.github.io/web-nfc/index.html#reading-nfc-tags are fine: they're non-normative and can use colloquial language.
1. For "a website must be visible and in focus", copy wording from http://www.w3.org/TR/page-visibility/, or just refer to its terms.
2. For 'only websites with a schema starting with "https://"', write it into the algorithms instead. https://webbluetoothcg.github.io/web-bluetooth/#requestDevice-secure-context has an example.

kenchris, 2015-08-18:
Thanks a lot for these pointers @jyasskin !

closed 2015-09-23 by zolkis