w3c-cg/web-nfc #40: Simplify process for obtaining permissions

alexshalamov, 2015-08-18:
At the moment spec defines that UA should obtain user permissions when:
- NFCAdapter.send() is called for 'tag' target
- NFCAdapter.send() is called for 'peer' target
- When event listener is added to NFCAdapter object (for listening to messages)

I couldn't find any other specs that require to obtain permission from user when event listener is added.

I would propose to have permission checks when:
- navigator.nfc.requestAdapter() is called
- NFCAdapter.send() is called (no matter if target was 'peer', 'tag' or 'any')

kenchris, 2015-08-18:
What does WebBluetooth do?

I think it is bad to check permission when calling requestAdapter because that will presumable be done in the initialization of the document, which in practise means up front permissions which we are trying to avoid.

annevk, 2015-08-18:
Adding event listeners must not have side effects. This is definitely the wrong design.

kenchris, 2015-08-18:
@annevk right, I also don't like prompting when adding listeners, but would it be possible to prompt before executing them? Basically prompt when receiving data (ie not share the data before) and potentially only prompt if there are listeners present?

annevk, 2015-08-18:
You mean on dispatch? That would be okay. Only prompting if there are listeners present would be less okay... Surely there must be some better design for this?

kenchris, 2015-08-18:
Yes on dispatch. The problem with requestAdapter here is that, no one expects to actually use it immediately after unlike the requestDevice (of WebBluetooth fame), because you never know when you might receive data - so developers will presumable call this as part of initialization. And we really want to avoid upfront prompts especially because it is hard for the user to associate that with an action

annevk, 2015-08-18:
Since the processing model for the event is not defined it's hard to come up with alternative models that might meet such a need. E.g., if it's always a one-off, using a promise-based pulling API might be better.

kenchris, 2015-08-18:
It's not. The browsing context will basically never know when someone bumps a tag or peer against the device.

annevk, 2015-08-18:
The adapter could have a `requestNFCData()` method or some such. That way there's also less of a chance of missing a message.

kenchris, 2015-08-18:
I guess then it might as well be a configuration on the requestAdapter (request read)

alexshalamov, 2015-08-18:
@kenchris WebBluetooth requires UA to inform user when requestDevice is called. https://webbluetoothcg.github.io/web-bluetooth/#security-and-privacy

alexshalamov, 2015-08-18:
@kenchris @zolkis Do you really want to have separate permissions for read / write? I think you could simplify it and have one "blanket permission" to access NFC functionality, otherwise, usability would be really bad, one popup after another.

If read/write permissions are combined, then we can leave permission check only for requestAdapter.

jyasskin, 2015-08-18:
I'd favor an `nfc.watch(...)` function that takes options about how the watch should proceed, and asks the UA to fire 'message' events when a matching NFC device is found. @sicking has a good point in #3 that if the user

1. opens a web page to example.com and
2. brings their device close to an example.com NFC tag/peer

that may be enough to infer that they've [expressed permission](http://w3c.github.io/web-nfc/index.html#dfn-expressed-permission), with no extra UA prompts. On the other hand, if their example.com web page calls `nfc.watch({url: 'https://other-site.com/path*'})` (maybe only in a future version of the API) then the UA might want to give users a choice when such a tag appears. UAs should be allowed to prompt on the `nfc.watch()` call, but I think they'd be unwise to do so.

I don't think the spec should narrow the "obtain permission" requirements to just [`requestAdapter`](http://w3c.github.io/web-nfc/index.html#widl-NFC-requestAdapter-Promise-NFCAdapter): let UAs experiment with exactly how they obtain permission. The current requirements do _allow_ UAs to prompt once at `requestAdapter()`, and infer permission thereafter, they just don't force it.

Spec-wise, the "[User agents must implement the following policies](http://w3c.github.io/web-nfc/index.html#security)" section doesn't work as-is: these restrictions need to be in the algorithms they apply to, in which place they can say what to do when the user hasn't given permission.

sicking, 2015-08-18:
Ah, yes, if there is API which allows a webpage to override what a device would normally do when a tag is tapped, then I agree that it might be useful for the UA to prompt at that point.

Though if what the device would have normally done is to navigate to the origin which initiated the override, then it doesn't seem like a prompt is needed.

It seems like very few of the APIs that are debated here is in the current spec draft. That makes it hard for me to provide feedback on them. But see my comment in #2 for why I think simple tag-reading/tag-writing/p2p-messaging does not need a prompt on tags that are explicitly marked as being readable/writable/messagable by the given website.

zolkis, 2015-08-18:
> I'd favor an nfc.watch(...) function that takes options about how the watch should proceed, and asks the UA to fire 'message' events when a matching NFC device is found.

I agree that having a watch() function instead of using addEventListener would be semantically more correct, especially for handling security dialogs and then optionally handle filters/constraints.
We used to have that function in earlier versions of the spec, but it seems we have oversimplified things and now have bumped into its price.

The current send() function is also semantically misleading, as in fact it means setting a message which will be sent when the proximity range is reached (an asynchronous operation with timeout).

> @sicking has a good point in #3 

That is fine with me too.

> Spec-wise, the "User agents must implement the following policies" section doesn't work as-is: these restrictions need to be in the algorithms they apply to, in which place they can say what to do when the user hasn't given permission. 

Ok, I will move them into algorithms.

kenchris, 2015-08-19:
Yeah, let's readd my watch method.

Maybe send() should be renamed to post() ? You know you post something but it might not be delivered - or at least not immediately ?

kenchris, 2019-02-26:
We currently only ask for permission during push, when the existing data on the tag doesn't contain a Web NFC record or the stored domain is different.

Closing for now

closed 2019-02-26 by kenchris