w3c-cg/web-nfc #49: Should NFCPushOptions.timeout be an implementation-dependent value or defined?

anssiko, 2015-09-16:
The group seem not to be fully settled on whether `NFCPushOptions.timeout` should have its value defined in the spec (e.g. 10 seconds as initially suggested), or whether this value should be implementation-dependent so that each implementation can adapt and use the underlying platform conventions and defaults what comes to the timeout for NFC operations.

I encourage folks to use this issue to gather use cases that would support either behavior, and consider that some platforms may have limitation with respect to what the lower and upper bounds for the timeout would be.

(Initial feedback from the list: https://lists.w3.org/Archives/Public/public-web-nfc/2015Sep/0052.html)

zolkis, 2015-09-16:
AFAIK there is no underlying platform convention and default wrt timers used with NFC on either of the platforms we target. Timeouts are only used with the Web NFC API, and the timers will likely be implemented on native side. Therefore a default value for timeouts is a matter of choice and it can be well standardized and be uniform across browsers. Since that number is arbitrary, it will always be a question why 10 seconds and not 8 seconds, but I would argue that a choice made is better than a choice left open for uncertainty.

The current API does indeed allow ```Infinity``` as a value for timeout. The spec does not specify what happens if the implementation cannot respect the value given by the developer.

I think we could include a Note saying that the value of ```Infinity``` is allowed (or not, votes pending).

Also, in the steps we should specify the behavior when the user agent cannot respect the values provided by developers for any reason, or specific reasons.

Note that the spec does not handle any corner case behavior e.g. when the platform would actually have hard-coded timeout with NFC operations that are smaller than developer-specified values. The API could include new method for fetching the platform dependent timeout value, and specify what happens in these conditions. So far this seems not a valid case. But should it occur later, the suggestion above should also cover this.

anssiko, 2015-09-16:
Calling out the meaning of `Infinity` explicitly would be good. It is used elsewhere on the Web Platform too to refer to a value that is unbound, and explicitly defined as such. Similarly, we could clarify the meaning of the value `0`.

Also as suggested by @zolkis we should clarify the failure bahavior if the UA cannot respect the timeout defined by the web developer. The note following this step should be clarified to explicitly say that out of bounds timeout value is another valid reason for termination at this step:

> An implementation may reject promise with "NotSupportedError", and abort these steps.

> NOTE
> The UA might terminate message push at this point. The reasons for terminations are implementation details. For example, the user could have has set a preference to allow a given origin only to read, write, or send data to peers. Also, the implementation might be unable to support the operation requested.

[1]: https://dvcs.w3.org/hg/dap/raw-file/default/battery/Overview.html

kenchris, 2015-09-16:
I don't see how using the value 0 is any worse than using the value 100ms. As we don't intent to special case all kinds of values like 100ms, I don't think it makes sense to special case 0 either.

zolkis, 2015-09-23:
#50 fixes this.

closed 2015-09-24 by zolkis