w3c-cg/web-nfc #51: Create per-origin NFC adapter properties section?

zolkis, 2015-09-21:
Like in the [Bluetooth spec](https://webbluetoothcg.github.io/web-bluetooth/#per-origin-device-properties)?

Or is the current way of specifying behavior in [algorithmic steps](http://rawgit.com/zolkis/web-nfc/gh-pages/index.html#dom-nfc-requestadapter) good enough?
Affected properties would be (per origin and per NFC adapter):
- suspended state (depends on visibility and focus of the Window of the Document of the script execution environment)
- push timer
- push buffers for tags and peers
- receive buffer.

jyasskin, 2015-09-21:
Are those properties shared between multiple execution environments within the same origin?

zolkis, 2015-09-21:
No, they are not. There are different suspended states, different timers, and push buffers. Receive buffer may be shared, but it's implementation detail. In fact all these are local private to implementations.

jyasskin, 2015-09-21:
Then these properties wouldn't go in a per-origin section. It might make sense to describe some of these (push and receive buffers) as [internal slots](https://webbluetoothcg.github.io/web-bluetooth/#terminology) of the adapter within the execution environment, and the suspended state could just come out of how the algorithms react to an unfocused browsing context.

zolkis, 2015-09-23:
#50 fixes this by introducing internal slots.

closed 2015-09-24 by zolkis