w3c-cg/web-nfc #52: IDL for NFCRecordData

riju, 2015-09-21:
From [1], it looks like NFCRecordData is changing from "any" to 
typedef (DOMString or unrestricted double or object or ArrayBuffer) NFCRecordData;

I had a quick look at the web-idl spec, http://heycam.github.io/webidl/#idl-union
and it looks like the above IDL is valid. But for Blink's IDL, at least as of now, Object is not supported as an union member.
I can raise the issue at https://code.google.com/p/chromium/issues/detail?id=240176

[1] https://github.com/w3c/web-nfc/pull/50/files#diff-eacf331f0ffc35d4b482f1d15a887d3bR1032

zolkis, 2015-09-21:
Should we switch back to ```any```? Or could you use ```any``` just in Blink instead of the explicit union (knowing this is an issue due to Blink implementation?). I'd suggest the latter.

riju, 2015-09-22:
We can do the latter, and when the Blink IDL issue is solved, switch back to NFCRecordData.

anssiko, 2015-09-22:
Given this appears to be an implementation related issue, I'll close this.

closed 2015-09-22 by anssiko