w3c-cg/web-nfc #53: How much needs to be hidden from pages that lose focus during NFC operations?

jyasskin, 2015-09-22:
The current spec allows pages to start operations in the foreground, and then leaks them some information if an event happens while they're unfocused/obscured (e.g. whether an NFC device came nearby). Does that match the security model folks are thinking about?

zolkis, 2015-09-23:
Pasting the context here.

Problem:
> jyasskin: This looks like an information leak: while the page is in the background, it learns the instant an NFC radio comes nearby. Instead, maybe just ignore any nearby devices until the page is focused again?

Reply:
> zolkis: In order to set up a watch, the page needs to get permission, and they do get a Promise. Then we need to respect the functionality: if the page is put in the background, there needs to be a feedback why the expected functionality (data transfer) does not happen. We can create an issue (i.e. needs more discussion) if you think this is not a good policy.

kenchris, 2019-08-05:
I see a few solutions.

1) Make sure that there is always a persistent notification that shows that a page is actively using NFC
2) Focus the page in case a read / write happens

@DonnaWuDongxia what happens today?

leonhsl, 2019-09-05:
https://w3c.github.io/web-nfc/#visible-and-focused-document
All NFC operations would be suspended when the page goes to background, what's the concern now?

beaufortfrancois, 2019-10-22:
I agree with @leonhsl. With all NFC operations suspended now, I think this issue can be closed. Am I missing something?

closed 2019-10-23 by kenchris