w3c-cg/web-nfc #54: cancelPush() should have target ?

riju, 2015-09-23:
At any moment, there can be a maximum of 2 pushMessage() for an adapter, 1 for tag, 1 for peer.
So correspondingly 2 messageBuffer/NFCMessage, 2 timers, etc. 
Presently in cancelPush(), all the instances of pushMessage() are invalidated. 
Do you want to add "target" as a parameter to cancelPush() ?

kenchris, 2015-09-23:
I think it is fine that it cancels everything for now. Cancelable Promises are on their way and we would want to migrate to using that when that happens (unless that is way in the future).

zolkis, 2015-09-23:
Until cancellable Promises come, we should add a target to cancelPush(). This would also be in line with having 2 separate timers, outboxes, and instances of the algo allowed.

zolkis, 2015-09-23:
Working on it now.

kenchris, 2015-09-23:
Please add a NOTE about this probably going away in the future

kenchris, 2015-09-23:
btw, some experimentation from domenic https://github.com/domenic/cancelable-promise/blob/master/test/cancelable-promise.js

zolkis, 2015-09-23:
#50 fixes this.

closed 2015-09-24 by zolkis