w3c-cg/web-nfc #60: TAG review

anssiko, 2015-10-06:
We should distill the W3C TAG review feedback when available (see the [Web NFC API w3ctag/spec-reviews issue][1]).

[1]: https://github.com/w3ctag/spec-reviews/issues/22#issuecomment-140744005

anssiko, 2015-10-15:
A work-in-progress draft [W3C TAG review feedback on NFC API][1] from @travisleithead of TAG (thanks!).

I suggest we distill this (still evolving) feedback at our joint [Web Bluetooth and Web NFC F2F meeting][2] at TPAC.

@travisleithead you're welcome to pop in at our TPAC F2F meeting if you have time on Monday Oct 26 between 15:00-18:00. Room 201, 2F in Sapporo Convention Center, see the [meeting wiki page][2] for further pointers.

[1]: https://github.com/w3ctag/spec-reviews/blob/master/2015/10/nfc-feedback.md
[2]: https://www.w3.org/wiki/TPAC/2015/ad-hoc-meetings#Web_Bluetooth_and_Web_NFC

kenchris, 2015-11-03:
We have added individual issues for the issues which were found in the TAG review and other which came up at TPAC.

OK to close this one @anssiko ?

anssiko, 2015-11-03:
I'll close this meta issue, since the TAG review feedback has now been split into [specific issues labeled with "TAG feedback"][1] that was discussed at F2F (also the [respective issue in w3ctag/spec-reviews][2] has been closed).

The specific "TAG feedback" issues are to be referenced in commits that address them to ease tracking.

[1]: https://github.com/w3c/web-nfc/labels/TAG%20feedback
[2]: https://github.com/w3ctag/spec-reviews/issues/22#issuecomment-140744005

closed 2015-11-03 by anssiko