w3c-cg/web-nfc #65: Consider using mimeType instead of type, type for kind

kenchris, 2015-10-15:
"The NFCRecord.type : this is the IANA media type, should jive with other uses of "mimeType" instead for clarity. NFCRecord.kind : this should likely be "type" instead."

https://github.com/w3ctag/spec-reviews/blob/master/2015/10/nfc-feedback.md

zolkis, 2015-10-15:
I fully agree with this. 
Let ```type``` become ```mediaType```, and  ```kind``` become ```type```.

IMO we should not use ```mimeType``` any more, then ```ianaMediaType``` could just be shortened as ```mediaType```.

kenchris, 2015-10-15:
I am fine with this too

zolkis, 2015-10-15:
I am implementing this change.

zolkis, 2015-10-15:
Should ```type``` be ```recordType``` for added clarity?

Now it looks a bit weird to have both ```type``` and ```mediaType``` in the same dictionary.

kenchris, 2015-10-15:
fine with me

commit 96fae5c6 (2015-11-03): Merge pull request #76 from zolkis/gh-pages

Fix #65. Improve requestAdapter() steps based on #67 discussion.
commit a5c4c7db (2015-11-03): Fix #65: change NFCRecord.kind to recordType and NFCRecord.type to mediaType.
closed 2015-11-03 by zolkis