w3c-cg/web-nfc #74: Potentially abstracting the payload contents a bit more from the NDEF records and from the Web NFC Records

kenchris, 2015-10-15:
https://github.com/w3ctag/spec-reviews/blob/master/2015/10/nfc-feedback.md

> NFCRecord and related low-level structures seems very integrated with NFC low-level protocols which they abstract. Not super "webby", but they do allow interop with existing NFC protcols and payloads not previously exposed via the web (in other words, there are some constraints). Even so, my feedback would be around potentially abstracting the payload contents a bit more from the NDEF records and from the Web NFC Records (while preserving the well-defined mapping that exists in the spec).

zolkis, 2015-10-15:
That would basically mean "guess the payload type from the JS type".
This is not good when developers want to control what kind of NFC records are being created, for the reason to be compatible with readers.
I think the current design strikes a good compromise between abstracting types, and controlling things on NFC side. 
I vote on not changing this now.

zolkis, 2015-11-02:
A concrete proposal related to this: #84

zolkis, 2015-11-06:
Fixed by #88.

closed 2015-11-06 by zolkis