w3c-cg/web-nfc #77: Provide option for suspending NFC watchers when "push a message" algorithm is not completed

alexshalamov, 2015-10-22:
Precondition:
There are active NFC watches (https://w3c.github.io/web-nfc/#the-watch-method) and "push a message" algorithm is not completed. UA is waiting for NFC device to be within proximity.

When NFC device (Tag) is within proximity, NFC watcher callback would be called and "push a message" algorithm would be completed, thus data will be written to a tag.

Problem:
If webpage has a list that is updated when device touches NFC tag and same page is trying to write data to a tag, it would be beneficial to suspend read operations during write.

Possible solution:
Provide option for pushMessage() that will suspend active NFC watches until "push a message" algorithm is completed.
Also, NFC watchers should not be called if same device is within proximity after "push a message" algorithm is completed.

zolkis, 2015-10-22:
So do you expect the spec to list normative policy for handling concurrent watch/push scenario?
Is there a policy that would be possible to implement across all platforms?
Otherwise should it be regarded as implementation detail?

alexshalamov, 2015-10-23:
@zolkis 
Spec may mention what happens when there are concurrent watch/push algorithms are running.
E.g. if nfc device comes within proximity: 1. Watchers are called, 2. Push message algorithm is completed.
It is implementable on linux, android, windows.

However, issue is about slightly different problem. 
Kenneth wanted to simplify life of developers by adding flag that will suspend watchers until push completes.

Something like adapter.pushMessage(... { target: "tag", ignoreRead: true })

With current API you have to do somethig like:

```javascript
var watchers = new Array();

function addWatchers() {
  navigator.nfc.requestAdapter().then((adapter) => { adapter.watch({}, (message)=>{addItemToCart(message);} ).then((id) => {watchers.push(id)} ) } );
}

function removeWatchers() {
  navigator.nfc.requestAdapter().then((adapter) => { watchers.forEach((id) => {adapter.unwatch(id)}) } );
}

function writeTag() {
   navigator.nfc.requestAdapter().then((adapter) => { removeWatchers(); adapter.pushMessage(....).then(() => {addWatchers();})} );
}

addWatchers();
=== user clicks write tag button ===
```
Another option is to have global flag:

```javascript
var pushPending = false;
navigator.nfc.requestAdapter().then((adapter) => { adapter.watch({}, (message)=>{ if(pushPending) return; addItemToCart(message);} )} );
navigator.nfc.requestAdapter().then((adapter) => { pushPending = true; adapter.pushMessage(....).then(pushPending = false;)} );
```

zolkis, 2015-10-23:
I like the flag idea, but not on push. IMO the watches should control this (their own) behavior. 
It should be a watch option something like ```skipReadOnPush```, with a default value ```true```.
Then we describe this in the watch steps.

zolkis, 2015-10-23:
We could also include a flag on push, like ```ignoreRead``` with default value ```false```, meaning to override watch behavior.
Then developers have 2 policies to choose from, they should cover all use cases.

kenchris, 2015-10-23:
I like the ignoreRead on push (that was my suggestion to Alex yesterday).

In an element world (like in Polymer) it is common to have separate elements dealing with read and write, so the current behavior makes it hard to do so, as you kind of need a global watch. That is of course solved if you can ignore the read.

Alex pointed out that the read can be useful if you want to cancel a write given the read values. I think that at least deserves an example in the spec, and maybe some discussion about the usefulness of that and the ease of use - if it is useful, could we make it easier?

zolkis, 2015-11-06:
Fixed by #88.

closed 2015-11-06 by zolkis