w3c-cg/web-nfc #79: Handle case when NFC support is disabled

alexshalamov, 2015-10-30:
Should developers be notified about case when NFC hw is disabled?
Some platforms only allow enabling NFC functionality from "settings app", no public API is available to enable HW.

Maybe "InvalidStateError" can be raised, so that developer can instruct user to enable NFC from settings application.

kenchris, 2015-11-03:
I actually don't think so. People disable the NFC themselves to save power etc, but then we just want to fail so they can enable it. We don't even want the developer to not show the NFC parts of their app UI

riju, 2015-11-03:
So we need to distinguish between "available" vs "present". The fingerprinting concern is that others can know that there is nfc. Lets add this error granularity later.

zolkis, 2015-11-03:
Currently the Promises just fail.
We could have a specific error for this, but one concern is fingerprinting (pages that get denied NFC access anyway could still sniff if there is NFC adapter on the device).

Since the two possible outcomes are
1. NFC not supported at all
2. NFC disabled,
the user agent could figure out what to do, without necessarily making the script aware.
So let's keep the current behavior of just returning a generic error.

kenchris, 2018-08-09:
@alexshalamov can we close?

closed 2018-12-04 by kenchris