w3c-cg/web-nfc #81: Clamp NFCPushOptions.timeout and provide default values

alexshalamov, 2015-10-30:
If NFCPushOptions.timeout can be set to Infinity, it should be clamped.

It will be also nice to have default values set for: 
NFCPushOptions.timeout to "0xFFFFFFFF max timeout"
NFCPushOptions.target to "tag" enumeration value

```javascript
dictionary NFCPushOptions {
    NFCPushTarget target = "tag";
    [Clamp] unsigned long timeout = 0xFFFFFFFF;
};
```

kenchris, 2015-11-03:
Why not just write a note that the UA might clamp to a UA specific upper bound?

riju, 2015-11-03:
Its ok to be explicit in the idl, geolocaltion, vibration, etc do it.

alexshalamov, 2015-11-03:
@zolkis 
http://www.ecma-international.org/ecma-262/5.1/#sec-8.5
http://www.w3.org/TR/WebIDL/#es-unsigned-long
If x > 2^32 − 1, then throw a TypeError

I wonder why other specs (MSE, WebAudio, Battery, DeviceLight) use "unrestricted double" for Infinity values? Should we also use:
```javascript
dictionary NFCPushOptions {
    NFCPushTarget target = "any";
    unrestricted double timeout; // default value +Infinity
};
```

zolkis, 2015-11-03:
Could we use ```unsigned long long``` and say the value 0 means no timer (i.e. timeout=```Infinity```)?
That would be in line with step 4 of http://www.w3.org/TR/WebIDL/#es-unsigned-long-long.

kenchris, 2015-11-03:
@anssiko your input please? :)

alexshalamov, 2015-11-03:
timeout: 0; means 0 ms, timer will be created and push operation would be cancelled in 0 ms.

I have to represent "infinity" value somehow in implementation, at the moment I use "unrestricted double timeout;" in the idl, which is mapped to "double".
```javascript
std::numeric_limits<double>::has_infinity => true
std::numeric_limits<unsigned long long>::has_infinity => false
std::numeric_limits<unsigned long>::has_infinity => false
```

zolkis, 2015-11-03:
OK, that solves it. Thanks.

PR #87 (2015-11-04): Change type of timeout to unrestricted double fix #81
Fix #81
commit 722ef2be (2015-11-04): Change type of timeout to unrestricted double fix #81
commit c3dafcb7 (2015-11-04): Merge pull request #87 from kenchris/gh-pages

Change type of timeout to unrestricted double fix #81
closed 2015-11-04 by zolkis