w3c-cg/web-nfc #82: Make NFCPushOptions optional for pushMessage

alexshalamov, 2015-10-30:
In my opinion, by making NFCPushOptions optional, we can improve API usability and developers would need to type less code for main use-cases.

```javascript
pushMessage(message);

// Instead of

pushMessage(message, {target: "tag", timeout: Infinity});
```

zolkis, 2015-10-30:
If we can have sensible defaults, why not.
I expect that most of the pushes will be done to peers, and tags are much more often read than written.
If we need to set default values, then I am inclined to make "peer" the default target for a push.

Timeout could be set to Infinity as a default.

zolkis, 2015-10-30:
We could also separate push (to peers) from write (to tags).
Then we could say
```
nfc.push(message1);
nfc.write(message2);
```

zolkis, 2015-11-06:
Fixed by #88.

closed 2015-11-06 by zolkis