w3c-cg/web-nfc #84: More flexible parameter to push()

zolkis, 2015-11-02:
Following up on a comment in the F2F review of Web NFC API, the ```push()``` method could use a simpler way to pass NFC content as argument.

The current way is to provide a sequence of NFCRecord objects.
```
nfc.push([{ kind: "text", data: "data" }], options);
```
We do need to keep the sequence, and the NFCRecords, for controlling the mapping to NDEF. 
To make the spec more coherent with the reads, we could modify it to provide an NFCMessage object, i.e. the sequence above + a URL path, and then deal with the same NFCMessage definition for push and reads.
```
nfc.push([{ data: { kind: "text", data: "data" }, url: "/mypath"} ], options);
```
But now this looks ugly. Still, some may say why to specify a dictionary with ```kind```, ```type``` and ```data``` properties. Even though when ```kind``` and ```type``` are not specified, the content is deduced from ```data``` type, if possible, and if not, it is treated as "application/octet-stream".

We could use defaults, and it would be fine to say:
```
nfc.push([{ data: { data: "data" }} ]), options);
```
But isn't that unintuitive, and still ugly?

So another possibility would be to define a union for ```push()``` parameter, which would simplify the most usual writes. Also, make sensible defaults to ```options``` so that it is possible to write:
```
nfc.push("mytext");
```

or 
```
nfc.push('{ prop1: 1, prop2: "hello" }');
```
The question is, what should be the sensible defaults.
IMO, we could say
```
typedef (DOMString or ArrayBuffer or NFCMessage) NFCPushMessage;
push(NFCPushMessage message, optional NFCPushOptions options);
```
Then we will have means to pass a simple data (e.g. serialized JSON as DOMString), and when needed, still have greater control on how the NDEF should look like. When DOMString or ArrayBuffer are passed, the resulting NDEF message will have 2 records (one for payload and one special record). If you need to pass more records, or specify more exact format, then use NFCMessage.

Thoughts?

alexshalamov, 2015-11-03:
+1 for simplification.
I've updated my idl to 
```javascript
typedef (NFCMessage or DOMString or ArrayBuffer) NFCPushMessage;
Promise<void> push (NFCPushMessage message, optional NFCPushOptions options);
```

kenchris, 2015-11-03:
Could you add some examples on how to write text, json, binary etc... then we should add such ones to the spec

zolkis, 2015-11-06:
Fixed by #88.

closed 2015-11-06 by zolkis