w3c-cg/web-nfc #86: add a cancelAll() instead of cancelPush()

kenchris, 2015-11-03:
This would probably suffice for most use-cases as devs can just readd the pushes, at least until we get cancellable promises. Also when that happens this method still makes sense.

zolkis, 2015-11-03:
We already agreed to make the parameter to unwatch and cancelPush optional, and if none provided, cancel all. IMO we don't need yet another method.

kenchris, 2015-11-03:
having it act differently when there are no parameters feels a bit like magic.

I was actually suggesting removing the other methods (which will become pointless when we get cancellable promises) and just add this instead.

zolkis, 2015-11-03:
We could call it ```cancelPush()``` and ```unwatch()``` without further ado. 
But then why not be able to pass an optional parameter if you only want to cancel one.
We arrive to the same thing from both ends.

kenchris, 2015-11-03:
Because that will presumable go away when we get better cancellation primitives in the platform. So I am getting to the conclusion that the limitation of not being able to cancel individual watches is not worth the future issues when there will be multiple ways of doing so.

I think that it is more a minor annoyance to not be able to cancel individual ones that actually a real issue. Also when we *do* get something like cancellable promises, there is still a use for being able to bulk cancel.

alexshalamov, 2015-11-03:
If we have "any" we can declare cancellation method as:
```javascript
Promise<void> cancelPush(NFCPushTarget target = "any");
```
@kenchris 
cancelAll - also cancels watches? :smile: 
I checked cancellable promises repo, I'm not sure how it will work for async operations that require UA to interact with underlying HW. Cancel would need to return another promise.

E.g. push().cancel().then(cancelled, cannot_cancel);
Another issue is that how do you cancel this: navigator.nfc.push().then();
Developers would need to cache returned promises everywhere, so that they can cancel unresolved promise when needed.

kenchris, 2015-11-03:
It should probably be called cancelAllPushes() or so

kenchris, 2015-11-03:
The other option people are exploring are CancellationTokens which would then basically be a special kind of Promise which you pass to the method.

```
CancellationToken t;
watch(..., t).then(...);
t.cancel();
```

Then watch would basically be a Promise.race between an internal promise and the cancellation token.

kenchris, 2015-11-03:
function watch(..., t) {
  let p = new Promise(...);
  ...
  return Promise.race(p, t);
}

kenchris, 2015-11-03:
@jakearchibald, I know you guys are looking into cancellable promises or cancellation tokens. We would really benefit from that work, but it seems that we have the requirement that the cancellation is also async, so it is basically a race between cancel and completion.

My suggestion is that we postpone individual cancellation for the Web NFC spec for now so that we can easily adopt these new features when they appear in the platform and just use a global cancelAllPushes() for now.

zolkis, 2015-11-03:
```CancellationToken```  seems to be an overkill for what we are trying the achieve here. Not even elegant any more (we wanted the change in order to get rid of a method). Now developers need to write half a page of code to replace a single function call.

IMO in this case it is better to apply the specialized semantics of ```cancelPush()```, rather than a generalized semantics (with all the issues) of cancellable ```Promise```.

I agree with @alexshalamov and
```javascript
Promise<void> cancelPush(NFCPushTarget target = "any");
``` 
works for me, although I am still on the opinion that it could also be a ```void cancelPush()```. We cannot guarantee cancellation (it is a race condition). We can only say that "we cancel if it's not too late for that". Eventually we can return "yes, we managed to cancel", or "sorry", but I am not sure about it. Anyway, if the Promise semantics can be ensured on all platforms, why not.

kenchris, 2015-11-03:
I was not suggesting people to write that, I was just trying to show what the idea behind a cancellation token was :-)

kenchris, 2018-08-09:
We have "cancellation tokens" in the Web Platform today. 

I created another issue for that (#147) so closing this.

closed 2018-08-09 by kenchris