w3c-cg/web-nfc #99: Wrong use of `SyntaxError`

zqzhang, 2015-12-22:
The 7th step of the push a message algorithm at https://w3c.github.io/web-nfc/#dfn-push-a-message says

> If the type of the message parameter is not DOMString or ArrayBuffer, and it is not an instance of NFCPushMessage, reject promise with "SyntaxError", and abort these steps.

Since here checks parameter's type, it shall reject promise with `"TypeError"`. Also per the WebIDL spec, here shall be  `"TypeError"`.

zqzhang, 2015-12-22:
Example: `navigator.nfc.push(() => {});` shall reject promise with `TypeError`.

alexshalamov, 2017-06-01:
@zolkis @kenchris This can be closed, I aligned 'SyntaxError' and 'TypeError' usage in API (I hope :smile:)

closed 2017-06-01 by kenchris