w3c-cg/web-nfc #102: Simplify the push algorithm to have one target at a time

zolkis, 2016-04-27:
The use case for writing a tag or pushing data to a device implies the following user interaction sequence with most available NFC applications:
1. The user selects the data to be transferred (e.g. file, picture, contact, WiFi network, etc)
2. The user agent prepares an NFC message that satisfies the use case (transferring a file, picture contact, or trigger connection to the specified WiFi network, etc)
3. The user selects the desired operation (write to a tag, or push to a device)
At this point NFC polling is started by the user agent for detecting tags or devices in proximity, and the NFC adapter must know the target: tag, device, or either.
4. The user taps the device to another NFC device. Usually a vibration happens to notify the proximity condition.
5. Optionally, the user taps again on the app screen to confirm the transfer.  The transfer takes place, and the app notifies about success or failure.
6. The user breaks proximity with the NFC device.

Given this sequence, the push algorithm could be simplified as follows:
- subsequent calls of the push() algorithm cancel the previous invocation and reject its promise with "AbortError".
- in step 11, run only one instance of the algorithm, with one slot containing the push message, and the target (either 'tag', 'peer' or 'any').
- cancelPush() doesn't need an argument any more, it cancels the currently pending push, except if the transfer is already in progress (device proximity detected).

This proposal would still allow all use cases to write to a tag or push to a device.
This proposal would simplify both the clients and the implementations by not having to deal with a parallel condition, whereas NFC push can only happen sequentially, both from end user point of view, and from the NFC adapter's point of view.

While the double slots could be reintroduced at a later version of the API if it proves necessary, it would be simpler and better to start with one slot. At this point I don't see any broken use cases with that.

kenchris, 2016-05-02:
I might be missing some things here.

My web app cannot know what the user is about to do, it touch tag or peer. It might have a more complicated flow when touching a peer. For instance, photo sharing.

- touches a tag - server generates URL for shared photo / album and just stores the URL
- touches a peer, actually sets up a connection (ie web socket or what not) and transfers the photos.

Both should work at the same time. How is this possible with your simplification? Maybe we should add such examples to the spec.

zolkis, 2016-05-02:
Your second bullet may not be the best example, but you're basically preparing a branch point in advance, depending on what target is tapped by the user, and you say the script needs to be ready for either.

I haven't seen use case like that so far, and doesn't seem to be wide-spread or high priority use case, especially for v1. It also has an uncertainty built in, and contemporary UI designs will almost certainly split this use case into two separate user interaction branches.

When a script implements an interaction use case, it almost certainly will be well defined with a given target, specifically for tags, specifically for peers, or the same data either to tags or peers, but I see very, very small likelihood for needing two parallel slots just in case.

But even when such parallel slots would be needed, why would they be classified by target only? You could have parallel slots based on other things as well, like script origin, permissions, etc. Then we'd need to update the spec with a more generic algorithm handling generic parallel push slots, exactly like how watches and dispatching are specified. 

I suggest we start with single push slot (tag, peer or any), and defer the rest for v2.

kenchris, 2016-05-02:
Then better create an issue for it and give examples on how to achieve this

zolkis, 2016-05-02:
This is the issue for it (could be labeled), and above I proposed a way to achieve this:
v1: simplify to support single push slot
[wait for feedback, collect and validate use cases for parallel push slots]
v2: support [more generic] parallel push slots.

It would be nice to hear people's opinion about this.

kenchris, 2019-02-26:
Can we close this @zolkis ?

zolkis, 2019-02-26:
What conclusion did we reach here?

kenchris, 2019-02-26:
> * subsequent calls of the push() algorithm cancel the previous invocation and reject its promise with "AbortError".

We have this:

> If target is "any", run the following steps twice, once with slot set to the value "tag", and once set to the value "peer"; otherwise run the following step once, with slot set to the value of target.
> If there are any existing instance of this algorithm running whose target is equal to slot, abort that instance of this algorithm by rejecting its p with "AbortError" DOMException.

> * in step 11, run only one instance of the algorithm, with one slot containing the push message, and the target (either 'tag', 'peer' or 'any').

See above quote

> * cancelPush() doesn't need an argument any more, it cancels the currently pending push, except if the transfer is already in progress (device proximity detected).

Doesn't exist anymore

> This proposal would still allow all use cases to write to a tag or push to a device.
> This proposal would simplify both the clients and the implementations by not having to deal with a parallel [...]

This would be as easy as change the above somehow, but isn't this more of a implementation simplification?

>

zolkis, 2019-02-26:
So by ```push()``` we mean _queue()_ with separate queues (of size 1) for tags and devices.
```any``` will fill both queues, otherwise only one queue gets updated.
Queues are cleared by a transfer, a next queuing, or window close.
Is that right?

kenchris, 2019-08-05:
I don't think this is worth it now. Very low priority if so, so closing for now

closed 2019-08-05 by kenchris