w3c-cg/web-nfc #104: Allow nfc.push to accept a dictionary

kenchris, 2016-05-12:
This way nfc.push({ name: "kenneth" }) should work

alexshalamov, 2016-05-13:
Now I remember why we don't have simple API for json data. NFCPushMessage is (NFCMessage or DOMString or ArrayBuffer). NFCMessage is Dictionary, so, you cant make NFCPushMessage to be (Dictionary or Dictionary or DOMString or ArrayBuffer). To fix that, we can make NFCMessage an object with constructor that takes NFCRecords.

zolkis, 2016-05-13:
Another way would be to separate the URL path from the records, in fact the path goes to the separate Web NFC record. IMHO the path could be included into the PushOptions or as a separate parameter.
The first parameter to ```push()``` would be the data in payload: 
- either a ```DOMString``` (text tag), or ```ArrayBuffer``` (media tag with ```application/octet-stream```,  or a dictionary (single object), as media tag with ```application/json```; in all these cases we push one record
- or, if it is instanceof ```Array```, then the elements should be ```NFCRecord```s.

Then, pushing JSON would be simple:
``` await nfc.push({ name: "ping", value: "pong" }, "/tabletennis/run");  // default push options```

Simplifying the single record use cases would be indeed important, since IMHO these would make up the majority of the API usages. Therefore I'd prefer this way to making NFCMessage constructible, although that's a valid option too.

closed 2019-02-15 by kenchris