w3c-cg/web-nfc #105: Bootstrap URL and WebBluetooth connection from tapping NFC tag

zolkis, 2016-09-19:
NFC supports handover to Bluetooth, plus loading URL from tag (similar to Physical Web). Having bluetooth.referringDevice already populated with the bluetooth connection would be great.

Examine what is needed to make this doable.

cyberphone, 2016-09-25:
I don't understand this use-case which may be due to a lack of details. You mean that a URL published by a Web page through NFC should:
- Make the invoking device setup an internet connection (access point) over Bluetooth via the device where the Web page is shown
- Open the invoking device's browser with this URL using the Internet connection

?

It is a pity that you haven't rather considered creating an NFC-invoked BLE channel to a _Web page_ since this would target a huge amount of already established high-volume real-world applications which currently are stuck with various inconvenient and security-broken OOB and QR schemes, _including 500 million Chinese users of mobile payment systems_.

cyberphone, 2016-09-28:
A related application which the entire industry is betting on is FIDO.  Although FIDO deals with NFC, they do not (AFAIK) target the "phone token" which is a de-facto standard.

A next step could cover a wide range of phone-app to pc-web interactions.  Payments + FIDO would help ensuring NFC support in future PCs (which may turn out to be the #1 problem).

cyberphone, 2016-10-09:
Highly related:
http://www.mobilepay.dk/da-dk/pages/The-story-in-English.aspx

cyberphone, 2017-03-22:
Is the Web NFC CG concluded?  There were never any explanation what this issue was asking for.  You should close issues that you consider invalid.

kenchris, 2017-03-22:
No, when tapping a device with NFC, the browser can auto connect to that device over Web Bluetooth.

The work on Web NFC spec has been a bit on pause as we get the level 1 features into Chrome and get developer feedback. That has finally happened and we are now trying to become part of the Origin Trial.

cyberphone, 2017-03-22:
@kenchris I'm still not clear what your proposal is with respect to the URL mentioned in the initial posting by @zolkis

Since the Hardware Based Security Services CG (initiated by Intel) didn't pan out, secure payments seem to continue relying on "Apps" which is why I'm suggesting a charter revision for a possible next phase.  The Web Payment WG (technically lead by Google) have had no "philosophical" problems with supporting native Apps including Android Pay.

The suggestion in a nutshell is allow Web NFC+ creating a paired BLE connection between a tapping device and the NFC-hosting Web page, including offering some kind of invocation attributes so that only specific Apps are started.  In this scenario it is rather the App and/or Mobile OS that is the gatekeeper which means that the Web page security context must be available as well. 

http://www.europeanpaymentscouncil.eu/index.cfm/knowledge-bank/epc-documents/epc-white-paper-on-mobile-payments-version-50/epc492-09-v50-white-paper-mobile-payments-edition-2017/
"For other mobile proximity payments (i.e. non-NFC based), the lack of standardization in the usage of the various proximity technologies (e.g., QR codes, BLE, etc.) is resulting in a very fragmented approach throughout Europe, and even at domestic level"

Actually even NFC payments are quite limited because they are currently based on card operations which fits squarely in mobile phones which are much more advanced than cards (like having a UI).