w3c-cg/web-nfc #107: Can webnfc access raw commands?

DeqingSun, 2016-11-21:
Hi.
I found found webnfc and I'm interested in having Android phone to access hardwares.

I have a battery-less board that I need to operate by accessing registers. On Android app I used NfcA.transceive

Is there an equivalent method in web?

Thanks

zolkis, 2016-11-21:
Implementation may use `transceive`. On the web API this is exposed as a `push(data, { target: "peer"})` invocation, and on the other side triggering a [`watch()`](https://w3c.github.io/web-nfc/#watching-for-content). Check examples [here](https://w3c.github.io/web-nfc/#examples).

Note that web-nfc is primarily meant for browser [use cases](https://w3c.github.io/web-nfc/#use-cases), so not really meant for low level access. For that you might look around on [npm](https://www.npmjs.com/search?q=nfc).

alexshalamov, 2017-06-01:
@zolkis @kenchris This is duplicate of #101 NfcA.tranceive is part of ISODep, can be closed.

kenchris, 2017-06-01:
Duplicate, closing

closed 2017-06-01 by kenchris