w3c-cg/web-nfc #110: Throw TypeError when invalid type is provided to nfc.push function

alexshalamov, 2017-04-24:
SyntaxError should be used when there is a parsing error, invalid syntax, tokens.

https://tc39.github.io/ecma262/#sec-native-error-types-used-in-this-standard-syntaxerror
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/SyntaxError

In cases when invlid type is provided to the function, TypeError should be thrown

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/TypeError
https://tc39.github.io/ecma262/#sec-native-error-types-used-in-this-standard-typeerror

Honry, 2017-05-02:
@alexshalamov , in section [10.7.1 The push() method]( https://w3c.github.io/web-nfc/#the-push-method), it still throws "SyntaxError" when NFCMessage is invalid, shell we fix them too?

e.g. 
>If record.data is not a string or number, throw a "SyntaxError" exception and abort these steps.

alexshalamov, 2017-05-02:
@Honry Do you refer to Mapping-to-Ndef section? NFCMessage to NDEF conversion requires parsing, validation of some fields, so SyntaxError should be fine there (IMO).

Honry, 2017-05-03:
OK, I see.

commit 54a4829a (2017-04-24): Throw TypeError when invalid type passed to push function

This PR changes exception code that is thrown by push function
when invalid type is provided.

Fixes #110