w3c-cg/web-nfc #114: Two syntax issues in Examples

Honry, 2017-04-24:
1. In Example 6, should use `${*}` instead of `$(*)`

> EXAMPLE 6: Push and read JSON (serialized and deserialized)

```console.log(`$(record.data.name) is $(article) $(record.data.title)`);```

2. In Example 8, the `data: utf8Text.buffer` is typeof ArrayBuffer, but the recordType is "text".
>EXAMPLE 8: Write Chinese text as UTF-8

```
navigator.nfc.push({ data: [
  {
    recordType: "text",
    mediaType: "text/plain; lang=zh; charset=UTF-8;",
    data: utf8Text.buffer
  }
]});
```

kenchris, 2018-08-09:
Right, do you feel like creating a PR?

Honry, 2018-08-10:
@kenchris. the first one had been fixed. 
And I just submitted https://github.com/w3c/web-nfc/pull/151 to fix the second issue, PTAL.

PR #151 (2018-08-10): Fix minor syntax issue in example
Fix #114
closed 2018-08-10 by kenchris