w3c-cg/web-nfc #128: "A Better Q"

cyberphone, 2017-06-05:
[image]

Using QR codes on the Web together with mobile phones acting as Identity Tokens or Wallets has been a huge success.

There is essentially only one snag; you need to start a specific QR- or QR-enabled application in order to use such a system.

However, if you scratch a bit under the surface of these systems you will find that they suffer from a fairly ugly security flaw: _There is no secure binding between the page showing the QR-code and the QR-code itself_. This fact has recently been successfully exploited by criminals who with simple phishing scams have lured people logging in to their bank for "Important Information" giving the phisher access to the account rather than the user. There is currently no publicly documented workaround for this vulnerability.

Note: even the most advanced systems out there using _Security Elements_ and _Asymmetric Key Cryptography_ exhibit this problem.

AFAICT, a _Dedicated_, _Write-only_ Web NFC variant could be a great replacement for inconvenient and security-broken QR schemes.  I don't see that a strict enforcement of the Web Security Model would be necessary since the Web service doesn't get any information from the user (via NFC).

anssiko, 2017-06-05:
@cyberphone, please give this issue a more descriptive name, such as "Proposal for a write-only Web NFC variant", and provide further technical details in this issue to allow the group understand and evaluate it better.

I don't see how QR codes are relevant to the technical discussion around Web NFC API. The scope of this group is defined in https://w3c.github.io/web-nfc/charter/#scope-of-work and discussion on QR codes is out of scope.

cyberphone, 2017-06-05:
@anssiko If the people reading this issue are unfamiliar with how QR-codes are used on the Web, there seems to be little point developing a detailed technical description.  BTW, I'm _not_ an expert on NFC.  A fruitful outcome would IMO require a _dialog_ between platform and application folks.  The people owning the aforementioned QR-powered systems usually do not talk to platform developers, making this task slightly challenging :)

The issue title is _deliberately_ somewhat "non-technical" because the same title and content have gotten quite a bunch of views (and 👍) by bank people on LinkenIn who I also hope will monitor possible progress on this topic _which already affects several hundred millions of users_.  An important part of this use case is that the same scheme with moderate adjustments would also be used in the physical (non-Web) world including POS terminals, ATMs, and Gas stations.

Now to the technical side of things _as seen from an application point of view_.  A QR-code provides "data" having no pre-defined function.  A Better QR using Web NFC would do the same, with one major difference: The security context (URL + Certificate) of the invoking Web page must also be available so that the OOB communication performed by the App in the phone can _prove_ where it got its invocation from to the server it typically calls.  There is typically "nonce" information involved as well but that belongs to the "data" provided by NFC.

Liryna, 2017-06-05:
>The security context (URL + Certificate) of the invoking Web page must also be available so that the OOB communication performed by the App in the phone can prove where it got its invocation from to the server it typically calls.

NFC is not made to be secure. Do not expect to use it for banking or any data that require secure transaction. Even with a context or unique data, the transaction can catch and replay by anyone.

alexshalamov, 2017-06-05:
@cyberphone 

>BTW, I'm not an expert on NFC.

Could you check whether [NFC Signature Record Type Definition (RTD)](https://en.wikipedia.org/wiki/Signature_Record_Type_Definition) would be sufficient for your use-cases?

cyberphone, 2017-06-05:
@Liryna wrote:
> NFC is not made to be secure. Do not expect to use it for banking or any data that require secure transaction. Even with a context or unique data, the transaction can catch and replay by anyone.

This is how most QR based schemes work today:

1. The user wants to login or have reached a Web page asking for a payment
2. The service offers a (preferably one-time) QR code in a Web page
3. The user starts a QR enabled application
4. The applications scans the QR code aided by the user doing the focusing
5. The QR enabled application performs the authentication or transaction using OOB communication based on the read QR data

An attacker standing behind the user or having a telescope can indeed take over the session after step 2.  NFC would make this harder but as you say, not impossible.  However, such an attack is (AFAICT...), not extremely useful, because it either authenticates the attacker or let him/her pay.

cyberphone, 2017-06-05:
@alexshalamov wrote:
> Could you check whether NFC Signature Record Type Definition (RTD) would be sufficient for your use-cases?

If signing data would be enough, signed QR data would do as well.

The core problem is that the QR applications this issue refers to must know which _page_ the information is emitted from in order to "simulate" in-bound communication. Using the NFC based QR replacement: If a page is phished, the phisher's security context will be supplied with the "data" which in the case of authentication will (in a properly designed service NB), generate an authentication that only works in the phisher's domain or be stopped already at the phone client level.

Liryna, 2017-06-05:
> or having a telescope

For NFC this would be a strong antenna. With a specific hardware, you can communicate to some meter with the device.

Why don't you look at a mutual authentication between the web page and a badge ? With this you would be sure that the communication is secured and that both device has the right to communicate to other eager without MITM ?

zolkis, 2017-06-05:
So you want to include a page id (origin/hash etc) in a read-only tag that guarantees integrity, so that any app can be sure the tag is associated with the given origin.
However, it is still vulnerable to MitM and replay attacks.
OOB channels are possible with devices, but not with tags.

cyberphone, 2017-06-05:
@zolkis wrote:

> So you want to include a page id (origin/hash etc) in a read-only tag that guarantees integrity, so that any app can be sure the tag is associated with the given origin.

Essentially that is it.

> However, it is still vulnerable to MitM and replay attacks.

Would it be possible describing a MitM attack?
Replay is thwarted at the application level using one-time challenges that are invalidated after being used.

> OOB channels are possible with devices, but not with tags

Right. See updated use case: https://github.com/w3c/web-nfc/issues/128#issuecomment-306182806

zolkis, 2017-06-05:
> Would it be possible describing a MitM attack?

It is possible to sniff NFC reads from up to 30 meters. Then it's a race condition. Also (read on below)...

>  Replay is thwarted at the application level using one-time challenges that are invalidated after being used.

The tags will still contain the same info, since we can't update the tags. Read it once, and you're as good as the tag itself, and on the copies of the tag you can change the info associated with the origin.

zolkis, 2017-06-05:
So I think we can limit this use case to devices.

cyberphone, 2017-06-05:
> It is possible to sniff NFC reads from up to 30 meters. Then it's a race condition

I believe I covered that in: https://github.com/w3c/web-nfc/issues/128#issuecomment-306182806

Anyway, this belongs to "Security Considerations" where _application developers_ should consider the effect of a possible session takeover.  In most current QR use-cases this does not create a security problem.  
Personally, I think this is an issue the _NFC Community_ should address since it is fully universal.

Regarding the other attacks which seem to be about malicious device software, this is out of scope for this proposal and most other security related specifications as well including TLS.  Signed data (a possible option) would invalidate modified tags.

Liryna, 2017-06-05:
> It is possible to sniff NFC reads from up to 30 meters. Then it's a race condition
I believe I covered that in: #128 (comment)

I do not see how this is covered.

>this belongs to "Security Considerations" where application developers should consider the effect of a possible session takeover

Developers do not use NFC for security purpose I believe.

>NFC Community should address since it is fully universal.

They do not do it probably because there are other standards made for this purpose (like ISO7816). Or if you are talking about reading distance, this is another layer level (hardware).

cyberphone, 2017-06-06:
> Developers do not use NFC for security purpose I believe.

@Liryna Right, but NFC is anyway used for security critical applications like EMV payments and e-Passport reading.  In order to use an unsecured channel like NFC for such purposes (including the ones covered by this proposal), you need an application specific protocol coping with some (or if possible all) imaginable attacks on the channel itself.

Just for my own education (I'm not an NFC expert you know):  Are there documented, "_long-distance_" attacks where the output from lets say a payment terminal, is not only read, but changed on-the-fly in order to trick the customer's card/device in some way?

I will _try_ to come up with a concrete security protocol for the targeted application space although the proposal even without that has considerably better security and usability properties than the current "gold standard", QR code.

cyberphone, 2017-06-08:
Regarding security and NFC: AFAICT, W3C WebAuthentication supports FIDO over the Web using a _dedicated_ NFC scheme.

My _initial_ _studies_ indicate that to succeed stealing a login from WebAuthentication using NFC as well as from a properly designed scheme based on this proposal, you need to phish the user and on-the-fly intercept and rewrite RF data.  More to come...

FREDFLT, 2017-06-12:
NFC and FIDO expert here. 

Few answers first about NFC:

"It is possible to sniff NFC reads from up to 30 meters.". 
=>NOPE. We are talking about ISO14443 here. Few centimeters max. 
Best known -non commercial- antennas can reach up to 25cm. 
Theoretically -never ever seen- you can reach 50cm.

"Even with a context or unique data, the transaction can be caught and replay by anyone."
=> You can choose secure protocols with encryption, mutual authentication and anti-replay attack. You are not forced to use non-secure protocols. (Yes few banks and payment services providers made obviously stupid choices and mistakes, mostly for back-compatibility issues)

Regarding FIDO:

If you are looking for a way to communicate with a NFC card/device (or a Bluetooth device) from a web page, you don't have to wait for W3C WebAuthN (FIDO v2), you can use FIDO U2F API today:
- Chrome for Android built-in support for many months -even if Authenticator app is still required for now- 
- Google play services API built-in support for Android Application development will come in the next few days/weeks, it was announced a few days ago at Google I/O '17
- Third Party iOS app SDK available but still complex to obtain for now (this may change "soon"(c) )

The first goal of FIDO U2F is to make secure authentication. If you are thinking about doing something else, this is still possible but you'll have to build your own protocol above U2F exchanges. Note: That's what I am working on with few other fools :)

cyberphone, 2017-06-13:
> If you are thinking about doing something else, this is still possible but you'll have to build your own protocol above U2F exchanges

Hi Fred, this proposal is _by design_ not tied to any specific application.  Some applications may on a high level overlap what FIDO does, some may do pretty different things.

The (from my horizon NB...), most obvious applications are mobile payment systems (aka "Wallets"), that unlike Apple, Google, Samsung, etc., do not build on emulating legacy "card" technology.   There are tons of such systems in use today and they are either based on QR scanning or that the user enters his/her mobile phone number.  That's stone-age :-)

FREDFLT, 2017-06-13:
Payment systems like Apple, Google, Samsung have their own problems.... and I REALLY don't care about doomed payment industry. We won't reinvent anything there. On the other hand, hardware wallet for cryptocurrencies, that's the future and we are working on it. 
BTW, I don't even understand why we are still talking about QR Codes here, this was not linked to Web NFC and QR code should only be used for one thing : easy retyping (copy/paste).

anssiko, 2017-06-13:
>I don't even understand why we are still talking about QR Codes here, this was not linked to Web NFC and QR code should only be used for one thing : easy retyping (copy/paste).

Let me repeat what I said already in https://github.com/w3c/web-nfc/issues/128#issuecomment-306130744:

The scope of this group is defined in https://w3c.github.io/web-nfc/charter/#scope-of-work and discussion on QR codes is out of scope.

-@anssiko, Web NFC Community Group Chair

cyberphone, 2017-06-13:
@anssiko @kenchris @zolkis @jyasskin If you feel that creating a viable NFC based alternative to the current "Gold Standard" for Web-2-Phone/App invocation is a bad idea you may close this issue now.

zolkis, 2017-06-13:
It's not very clear how your proposal would affect the v2 API.
What we are considering is @alexshalamov [suggestion](https://github.com/w3c/web-nfc/issues/101#issuecomment-304653157), but not at this time.

cyberphone, 2017-06-13:
> It's not very clear how your proposal would affect the v2 API

The proposal isn't associated with any particular API version.  At this early stage I only need to know if you consider the application _potentially_ interesting and useful.  The proposal has AFAICT no relationship to the suggestion https://github.com/w3c/web-nfc/issues/101#issuecomment-304653157 by @alexshalamov since ISODep doesn't deal with Web security contexts.

cyberphone, 2017-06-13:
@FREDFLT The _intention_ with the proposal was creating a technically and politically "neutral" interface that (for example) would enable wallets supporting entirely different types of payment instruments.

Anyway, since every bank out there is busy implementing QR support, we will soon be one big happy family :-)

FREDFLT, 2017-06-13:
*Yawning* I don't care about QR Codes. I don't care about banks. This "issue" has no connection with Web NFC.

anssiko, 2017-06-14:
There seems not to be wider interest in this group to pursue this proposal further at this time, so I'll close this issue now.

closed 2017-06-14 by anssiko

cyberphone, 2017-06-14:
> There seems not to be wider interest in this group to pursue this proposal further at this time, so I'll close this issue now.

Understood.  This group prefers targeting new applications with very limited scope and/or established applications which since long time back are served by other solutions.

cyberphone, 2017-06-15:
Although the use case has been dismissed, I promised a more complete description for review.

[image]

Assumption: _The Service, PC, and Phone are free from malware interfering with the devised scheme_.

The security of this scheme is based on multiple factors:

- Public key cryptography exposes no static secrets to attackers
- One-time challenges limit attacks to the specfic session
- Session cookies, only known by the Service and the user's PC (Browser), render intercepted NFC or authentication objects useless outside of the user's PC
- Intercepting and rewriting RF data on-the-fly appears to be quite difficult
- The Web Security context provided by the NFC solution in conjunction with signing thwarts basic "phishing" attacks
- The user must perform an action in order to authorize a login

The original (and possibly updated) document is available at: https://cyberphone.github.io/doc/research/nfc-based-qr-replacement.pdf

FREDFLT, 2017-06-15:
"NFC in PC" => doesn't exist. End of story.

cyberphone, 2017-06-15:
@FREDFLT Is this a "terminology" issue or is it about the current state of NFC support in PCs?  I can consider updating the text if it fits into the already pretty crammed document :-)

FREDFLT, 2017-06-15:
99.999% PCs don't have NFC readers. If you installed a NFC reader on a
computer, OS and application behaviors will be completely different if you
got Windows 7 or 10, OS X, Linux... but let's focus on Windows. Having this
PC access to "Web NFC API" through a peer device... man, this won't
happen... but whatever, I don't even believe in this WebNFC initiative, so
I will stop the discussion here and focus on existing technologies and
plausible use cases. If you want to continue discussion outside WebNFC
scope, you already got my mail. :)

cyberphone, 2017-06-15:
> 99.999% PCs don't have NFC readers

Although this number may be slightly exaggerated, this particular issue (which I have mentioned on multiple occasions), is of _no concern to Intel_ who's primary objective (based on https://w3c.github.io/web-nfc/use-cases.html, earlier list conversations, and a Skype conference), is getting rid of _perfectly working_ "Apps" in mobile devices like e-passport readers.

High-volume, consumer-centric, security-critical applications are definitely out of scope.

> I don't even believe in this WebNFC initiative

Me neither, Web NFC is a classic "_Solution Looking for a Problem_"

> focus on existing technologies and plausible use cases

Since my "raison d'être"  is designing crazy stuff nobody * wants, I guess will continue with that a while :-)

*) The Alibaba folks claims that QR is what the world want, not NFC