w3c-cg/web-nfc #131: Payments - An Invalid Use Case

cyberphone, 2017-06-17:
If you read this document
http://w3c.github.io/web-nfc/index.html#payment-scenarios
you can get the impression that a future revision could support payments.

This is incorrect, NFC based payments are currently based on low-level card protocols that were _not designed for the Web_.  In addition, Mobile "Wallets" are exclusively based on native applications.  It will stay like that since there is no "AndroidKeystore" for the Web.  Efforts for creating such features have been going on since 2011 but have to date always failed, most recently in https://www.w3.org/community/hb-secure-services/

The only way Web NFC could ever support payments would be through entirely new schemes but no such activity seems to be in the workings.

zolkis, 2017-06-17:
That use case is not for payment itself, but recording the fact of payment with a website in order to have secondary actions. However, we do need to revisit that section.

cyberphone, 2017-06-18:
Well, if you "read between the lines", this use case as well as most of the other listed use cases are really about providing Web based alternatives to native "Apps".

Although I absolutely love the Web, there are plenty of applications out there that are better served by "Apps".  For security related applications like payments you need signed and vetted applications which doesn't have a Web counterpart.  However, a viable (and extremely popular), way forward is _combining_ the Web and App worlds since these worlds offer unique capabilities of their own.

The recent W3C Web Payment API is the first Web standard of this kind, offering an application specific "bridge" between the two worlds, currently running on Android (for Android Pay and similar).  This API will presumably be extended to cope with whatever additional features are needed including loyalty cards.

Outside of the W3C, Mozilla and Microsoft are now shipping emulations of Chrome's "Native Messaging" which is a variant on the same theme.

It had been pretty cool making a Web NFC bridge but that would require an entirely different project since not even the required hardware is available: https://github.com/w3c/web-nfc/issues/128#issuecomment-308838885.

zolkis, 2017-06-18:
> this use case as well as most of the other listed use cases are really about providing Web based alternatives to native "Apps".

Actually quite the contrary. This group intended to work on Web-specific use cases of NFC (as a transport/interaction mechanism) rather than providing a generic NFC API for making Web based alternatives to native apps. This is also the reason for departure from the lower level APIs of the former NFC WG. And again, the payment scenarios did no mean to _do_ payments with Web NFC, but rather, do *something else* when an NFC payment is done.

cyberphone, 2017-06-18:
>> this use case as well as most of the other listed use cases are really about providing Web based alternatives to native "Apps".

> Actually quite the contrary. This group intended to work on Web-specific use cases of NFC (as a transport/interaction mechanism) rather than providing a generic NFC API for making Web based alternatives to native apps.

I think you are making things more complicated than they actually are :-)  If we stick to: https://w3c.github.io/web-nfc/use-cases.html, IMO, most (if not all), of these applications can be performed without Web NFC using Apps + NFC or NFC with URL + Browser.

I may reuse your work if you get it to run on Windows.  It should be pretty simple creating the Web NFC scheme I outlined.

anssiko, 2017-06-19:
@zolkis, could you take a stab at clarifying http://w3c.github.io/web-nfc/index.html#payment-scenarios along the lines of https://github.com/w3c/web-nfc/issues/131#issuecomment-309198225?

cyberphone, 2017-06-20:
In have difficulties understanding the motives for running an NFC based standardization effort that (AFAICT) doesn't address any of the existing high volume NFC use cases, as well as not adding (AFAICT) a single new feature that you can't do with existing technologies and practices.

The lack of a PCs solution and the unlikely buy-in from Apple, in reality makes this a Chrome-for-Android-specific scheme which seems more like a Google/AOSP project than a W3C activity.

If any of the claims above are wrong, please update relevant documents so that the capabilities and objectives become clearer!

If you want to develop full-featured NFC applications in a Web-like environment, you can actually do that right now using Android's excellent WebView which doesn't suffer from the same limitations as the "true" Web: https://cyberphone.github.io/doc/saturn

anssiko, 2017-06-20:
LGTM with a comment.

anssiko, 2017-06-20:
Fixed with #132.

closed 2017-06-20 by anssiko