w3c-cg/web-nfc #138: Read of NFC tag without Web NFC Record

OlivierLorente, 2017-12-13:
Hi,

We have tried the nfc.watch() and nfc.push() interfaces to read or write a NFC tag on an Android phone. It is working fine: We can write a Web NFC Message containing a URI record + a Web NFC record. We can then program a “watch” action and read the content of the tag when we tap it.

When we setup a watch, we have noticed that it works fine if the tag contains a Web NFC Message but it doesn’t work if the tag contains an ordinary NDEF message (without Web NFC record). Is that intended? I guess that it is intended for security reason because you use this Web NFC Record to know if you can trust the content of the tag. Is it correct?

In my opinion this is a big limitation because the NFC Web App can only read tags that have been programmed by another Web App.
Can you please confirm this behavior?

Thank you, regards
Olivier

kenchris, 2017-12-13:
Try changing mode (https://w3c.github.io/web-nfc/#dom-nfcwatchmode) to "any"

OlivierLorente, 2017-12-13:
Would you have an example showing how to use the nfcWatchOption ?

I have tried the code below but I get a "TypeError".

I'm not sure if I have used the correct value for the fields url, recordType and mediaType. I have let them empty but the documentation says: "The default value "" means that no matching happens".
If my understanding is correct, if I let those fields empty, no match will happen so I will not receive any notification.

What shall I use to be notified whatever the  url, recordType and mediaType present in the tag? I set this nfcWatchOption only for the mode, I don't want to put some filtering conditions on the other ones.

Thank you

```
function readNfc() {
  if ('nfc' in navigator) {
    var nfcWatchOption = {
        url: "",
        recordType: "",
        mediaType: "",
        mode: "any"
      };

    navigator.nfc.watch(function(message) {
        processMessage(message);
      }, nfcWatchOption)
      .then(() => consoleLog("Added a watch."))
      .catch(err => consoleLog("Adding watch failed: " + err.name));
  } else {
    alert('NFC API not supported.');
  }
}
```

OlivierLorente, 2017-12-13:
I hadn't seen that you have edited your answer. I'm going to try that. Thanks

`

> NFC.watch(callback, { mode: "any" })

`

kenchris, 2017-12-13:
Don't set things to empty string "" for things you don't need :-) just leave it out

```
    var nfcWatchOption = {
        url: "",
        recordType: "",
        mediaType: "",
        mode: "any"
      };
```

```
const options = { mode: "any" };
navigator.nfc.watch(message => processMessage(message), options);
```

closed 2018-08-09 by kenchris