w3c-cg/web-nfc #139: Unable to transfer data between peer-devices

MartinBeekhuizen, 2017-12-29:
Hello,

I did some experiments with the web-nfc API and I'm curious about why the peer-to-peer examples does not work.

The API works fine when pushing and reading messages to and from passive NFC tags.
However when trying to push and read a message from a peer to peer device it does not work.
I tried setting the nfc push options to 'peer' and 'any' and provided a 'timeout' but without success.
So after waiting for the x milliseconds provided the timeout error kicks in.
It looks like the devices are unable to find each other.
The devices are able to transfer data to each other when using the android beaming service.

Is this behavior when reading and pushing data between peer devices known?

Any guidance on how to push/read data to/from peer devices would be appreciated.

jvdassen, 2017-12-30:
What browser did you use to test this? I assume Google Chrome?

I tried the same and it raises a `NotSupportedError` when I'm explicitly targeting `"peer"`. Looking at step four in the [specs](https://w3c.github.io/web-nfc/#steps-push) for the push operation, I noticed that it said:
> An implementation MAY reject promise with "NotSupportedError" and abort these steps.

That would indicate to me that pushing to peer targets is not supported / implemented by Google Chrome at the moment.

kenchris, 2017-12-30:
I can confirm that it is not implemented, mostly due to it requiring Android native UI elements at a platform level.

MartinBeekhuizen, 2017-12-30:
@jvdassen You assumed correct. I too experienced the same error.

Thanks for confirming the issue @kenchris .

Is there any news on when developers can expect the peer function to work?
Or when the Web NFC API is being released to the public?

kenchris, 2017-12-30:
We are planning at attempting to add an origin trial for  this after we ship Generic Sensor APIs

@anssik @scheib

jvdassen, 2018-01-02:
@MartinBeekhuizen If you need a workaround to transfer data between peer devices from the browser, you could leverage the [Web Share API](https://developers.google.com/web/updates/2016/09/navigator-share), to pass the data over Android Beam. Depending on the data you are trying to send, this might not be a solution for you; in my case this worked perfectly since i tried to send an URI, which should be opened in the target's Browser.

mrj, 2018-05-08:
Is there an estimate when peer-to-peer URL transmission by WebNFC will be implemented in Chrome on Android? I'm not having success on Version 66 with
```javascript
navigator.nfc.push({records: [{recordType: 'url', data: url}]});
```
The suggested use of the Web Share API doesn't work for arbitrary URLs, only for the URL of the page itself.

Would it help to add a comment to the [Chromium Bug](https://bugs.chromium.org/p/chromium/issues/detail?id=520391) that states that peer-to-peer is not yet implemented?

alexshalamov, 2018-05-08:
@mrj Chromium [bug](https://bugs.chromium.org/p/chromium/issues/detail?id=520391) that you mentioned is about whole WebNFC feature and UA developers can decide not to support P2P if needed. 

You could add comment to the bug that P2P feature is desirable. Also, you can check [design document](https://docs.google.com/document/d/1-pfde7lzZqNQXdu1JkxWBywvG_fVcrmljdPbG99oHPY/edit#heading=h.430ahftk4vy) for the feature, I've implemented P2P support, however, it was decided to postpone it.

mrj, 2018-05-08:
Thank you Alex. I've read your linked document, patch, and about Android Intents.

I still don't understand why code is needed to support P2P. If WebNFC can currently write a URL to a passive tag, and a passive tag with a URL can currently trigger a browser selector or direct navigation on Android, what is Android Chrome WebNFC sending differently to a passive tag such that the receiving Android device doesn't respond? Is a lot of coding involved? Was the postponement due to the amount of work to polish and release, an assessment of the demand for and the utility of this feature, or corporate strategy?

alexshalamov, 2018-05-08:
@mrj Android Beam™ is used to exchange messages (opened tab url) between active devices, in my patch, I had to modify codepath in a way that will override Chrome's default behavior if WebNFC message is pushed / watched for 'peer' or 'any' target (some might consider it as an API / UX break).

Anyways, if you have proposals, concerns or feature requests related to Chromium, please file an issue or add comment in Chromium [bug](https://bugs.chromium.org/p/chromium/issues/detail?id=520391) tracker. WebNFC github issue tracker is not an optimal place to discuss UA specific implementation details.

kenchris, 2020-04-22:
SNEP (peer to peer protocol is deprecated on Android and not supported on Apple OS'es).

For this reason we cannot support this feature. The TNEP and HCE provides a kind of replacement

https://github.com/w3c/web-nfc/issues/559

closed 2020-04-22 by kenchris