w3c-cg/web-nfc #140: Secure URL from Web Page

cyberphone, 2018-01-29:
This is a variant of what I have previously proposed: https://github.com/w3c/web-nfc/issues/128

This idea is though much simpler though it is only about retrieving a secured URL from a Web page.

Scenario: A Web page wants to present a URL through NFC.  Only if the Web page and the URL to be presented belong to the same domain, NFC will actually emit the URL.

My belief is that this can support entirely "phish-free" OOB authentication schemes, but I have yet to perform a full-blown analysis.

mrj, 2018-03-16:
If peer-to-peer URL transmission by WebNFC is to be restricted to match the domain of the sending site, I'd also allow transmission to subdomains, and allow relative URLs to be used in WebNFC `push` calls.

cyberphone, 2018-07-09:
As a picture:

[image]

With more detail: https://github.com/cyberphone/qr-replacement

cyberphone, 2018-07-13:
### Possible "Side Effects"

Although uniting payment protocols is not a part of this proposal, it is related since the idea is that somewhere down the line, it should be possible using the same payment protocols in physical shops as on-line.

NFC would in such a use case only be used to setup the communication channel while the actual payment protocol would use HTTP or WebSocket.

[image]
In such a setup there is no need for a traditional payment terminal since the entire client side of the
UI and security is catered for by the mobile device.

kenchris, 2020-05-20:
This is outside our scope for now

closed 2020-05-20 by kenchris