w3c-cg/web-nfc #142: Use cases for emulating passive NDEF tags

leolux, 2018-03-28:
Would it be possible to bring HCE to the web and emulate NFC Forum compliant tags?

One of many use cases would be access control in cinemas: Users can buy cinema tickets online and use their smartphone to check-in in the cinema area by touching a NFC reader device without having to install an native app.

cyberphone, 2018-03-28:
Is it possible describing this use case in a step-by-step-fashion?

cyberphone, 2018-04-15:
@TMesot It it [quite] possible that I completely misunderstand the plot, but it seems that the ticket page is on a PC (in most cases).  This use case has already been dismissed and PC vendors have [accordingly] ceased supporting NFC as well.

leolux, 2018-04-15:
My idea for the HCE use case looks a bit different:

1. Go on your ticket page (installable PWA) with your mobile browser (no PC involved at all)
2. Search for the event your want a ticket for
3. Go to checkout
4. Login with your favourite auth method (email, facebook, etc.) which create a user id
5. Buy the ticket
6. After payment transaction has completed the webserver behind the ticket page generates a ticket id and adds this ticket id to the list of tickets for this user id within its own database
8. Later on the user taps a NFC tag or scans a QR code at the entrance of the event to open a event specific webpage. Or the user opens the ticket PWA from its home screen and clicks on a button of the current event which enters the same event specific webpage
9. The event specific webpage tells to user to stay on this page during NFC checkin and the page uses the WebNFC API to implement the HCE used in the following step
10. The user taps on a NFC reader/termnial which reads the user id from the emulated passive tag
11.. The terminal connects to the ticket platform (via internet) and checks if the given user id has a valid ticket id

The basic idea is as long as the current webpage is open and in the foreground the operation system can use the HCE implementation of tthis webpage.

TMesot, 2018-04-20:
@cyberphone 
The idea is to buy your ticket with PC or smartphone and then, use the smartphone at entrance for validation. We currently use QR Code inside PDF (send by email) or with dedicated page under "My account" part of web site after login showing the QR Code. The idea is on this specific page, the ticket ID encoded inside the QRCode can also be broadcasted via NFC Write. It's the same need as step 8 of @leolux 

For us the HCE is not needed in our use case, we don't need the security needed by smartcard. The goal is only to broadcast an unique id which is validated server side by the NFC reader. QRCode works fine but you need people increase screen brightness and a good camera. With NFC, you could be able to do it whatever the environment condition are.

cyberphone, 2018-04-24:
@TMesot @lcarcone The WebAuthentication folks spent years on creating a "Smart card for the Web".  Probably for a reason.

Related: http://www.atimes.com/article/china-launch-e-id-cards-citizens-via-phone-qr-codes/

thelamer, 2020-06-12:
We just implemented NFC scanning for our Web Wallet here https://github.com/linuxserver/nano-wallet/ (thanks for all the work in the web-nfc project)

Given the major use case of people transferring funds phone to phone using insecure text data ( in this case just a string containing the destination address and amount) being able to broadcast a tag from the same interface as well as scan them is extremely useful if planning to completely replace QR codes with NFC for the transaction workflow. 
This allows people to simply bump phones instead of scanning a QR presented on one phone with their camera. 

If you are not familiar with the process or other payment platforms like WeChat pay you can see a basic demo here: 
https://twitter.com/garrytan/status/966122234271051776?lang=en