w3c-cg/web-nfc #152: Use addEventListener instead of watch

kenchris, 2018-09-27:
I don't recall why we invented the watch, but it feels a bit like its own thing, instead of being aligned with the platform. We could go with a `startNotifications()` approach like WebBluetooth.

```js
navigator.nfc.addEventListener('exchange', message => {
  for (let record of message.records) {
    let article =/[aeio]/.test(record.data.title) ? "an" : "a";
    console.log(`${record.data.name} is ${article} ${record.data.title}`);
  }
},{
  url: document.baseURI,
  recordType: "json"
});

navigator.nfc.startNotifications();
```

zolkis, 2018-09-27:
We could add a filter to `watch()`, i.e. options.
It would be nice if `addEventListener()` supported options.
Next possibility is Observables (not that it would be aligned with the platform, either).
Or just do away with options.

kenchris, 2018-09-27:
addEventListener does support options today :-)

zolkis, 2018-09-27:
But not _those_ NFC options. Is it better to extend a standard dictionary with NFC-specific properties (making it _apparently_ aligned), or have a separate call with its own, admittedly different semantics?

kenchris, 2018-09-27:
I think it is better extending. Creating something that looks similar but has edge cases that works differently isn't really good.

Other option would be (feel free to bikeshed names)

```js
const filter = navigator.nfc.createFilter({
  url: document.baseURI,
  recordType: "json"
});

filter.addEventListener('exchange', message => {
  for (let record of message.records) {
    let article =/[aeio]/.test(record.data.title) ? "an" : "a";
    console.log(`${record.data.name} is ${article} ${record.data.title}`);
  }
},{
  once: true
});

filter.startNotifications();
```

zolkis, 2018-09-27:
> I think it is better extending. 

Maybe so, but what is the general consensus or mechanism on extending standard dictionaries in the platform? Do we need to register those somewhere? (imagine multiple specs adding properties which may clash)

> Creating something that looks similar but has edge cases that works differently isn't really good.

Extending a standard dictionary looks exactly like this. So it's a counter-argument.

> `const filter = navigator.nfc.createReadFilter({`

So we'd need to specify a new NFCFilter interface that contains the `startNotifications()` method? 
Isn't that more complex than just using a `watch()` method?

One way we could do it is define NfcWatchOptions and NfcPushOptions as interfaces (strip the Options from the name), add the `startNotifications()` method, that would return an object that has the `cancel()` method.

kenchris, 2018-09-27:
It's more similar to WebBluetooth which has `getCharacteristic()` which returns an object with `addEventListener` and `startNotifications()`

zolkis, 2018-09-27:
On the other hand we can't really expect to handle NFC interactions with Bluetooth interactions.

reillyeon, 2018-09-27:
I think in general APIs are migrating away from explicitly passing callbacks in favor of either Promises for one-shot events or event listeners for repeating events. I like the idea of creating an `NFCWatch` (or `NFCReadFilter`) and adding an event listener to that. Unless it is absolutely necessary a `startNotifications()` method could be overkill. It is useful for Web Bluetooth because turning on notifications requires communicating with the device and so can fail in more interesting ways. `addEventListener()`/`removeEventListener()` could be sufficient for turning on and off notifications.

zolkis, 2018-09-27:
Yes, something like this would do:
```javascript
[Constructor(NFCWatchOptions options), SecureContext]
interface NFCReadFilter: EventTarget {
  readonly attribute USVString url;
  readonly attribute NFCRecordType recordType;
  readonly attribute USVString mediaType;
  readonly attribute NFCWatchMode mode;
  attribute EventHandler onread;
};

interface NFCReadEvent: Event {
  readonly attribute NFCMessage message;
};
```

or if we want to bind it to the `navigator.nfc` object, we could have a factory method instead of constructor (like @kenchris suggested).

kenchris, 2018-09-28:
Yes, that is a fear, so we should be careful when doing so, and work with the people who specced the upstream feature.

Luckily, we also pass these specs through the TAG review which should catch most of these.

With the `NFCReadFilter` we are not really doing that though.

For factory function, I think that would make more sense for the NFC object if we expected to have multiple of  those, and would be in line with Generic Sensors. But it is not really a sensor and we don't expect to have multiple (if so a default could be chosen by the browser, I guess, like camera)

zolkis, 2018-09-28:
I am fine with factory methods, but if we wanted constructors it's well possible.

Filters are independent of NFC adapters, so no real need to bind to the nfc object syntactically (only in runtime: when when filters are constructed, implementations/algorithms should do the standard checks for support, secure context/permission etc and throw if there is an issue).

Even if there are multiple NFC readers, we have agreed to have an additional API in the future for enumerating/selecting them for write (currently all of them can read, and any of them is allowed to write, so if they are too close, it might happen to attempt writing a tag from two adapters, which is anyway a deployment problem and should be avoided). So at any point implementations will know how to handle the NFC adapters.

However, the API would be perhaps cleaner with the filter creation under the nfc object, since after this change it would only contain the `push()` (to become `write()`?) method. I suggest keeping the `watch()` name for a factory method of creating read filters. 

```javascript
[SecureContext]
interface NFC {
  Promise<void> push(NFCPushMessage message, optional NFCPushOptions options);
  Promise<NFCReadFilter> watch(optional NFCWatchOptions options);
};
```

Note that `NFCPushOptions` and `NFCWatchOptions` may now include an `AbortSignal` bound to an `AbortController`. The limitatation is to be able to cancel a push only in the way it was created (not like create with `any` and cancel only for peer push).

domenic, 2018-09-28:
You shouldn't modify a foundational method like that of the platform for NFC-specific use cases. So the "Other option" from https://github.com/w3c/web-nfc/issues/152#issuecomment-425045567 makes more sense to me. (Although I'm unsure why you'd need to do `{ capture: false }`, since it's not in a DOM tree anyway.)

Otherwise, I strongly agree that inventing your own event listener-like infrastructure is not good, and following the precedent set by other specs makes more sense. Especially if we already have such a strong parallel with Web Bluetooth.

> or if we want to bind it to the navigator.nfc object, we could have a factory method instead of constructor (like @kenchris suggested).

Constructor is preferable if possible! I wonder why Web Bluetooth went with something else. Does creating it have to be async? Your last IDL block has it as promise-returning, and it's not clear to me why.

reillyeon, 2018-09-28:
> Constructor is preferable if possible! I wonder why Web Bluetooth went with something else. Does creating it have to be async? Your last IDL block has it as promise-returning, and it's not clear to me why.

`BluetoothRemoteGATTCharacteristic` does not have a constructor because you have to interrogate the device in order to determine what characteristics are available. I generally support using constructors however it becomes awkward to use when you can construct an object representing a capability that doesn't exist. Then you need a mechanism to report that the object is invalid rather than refusing to create it in the first place.

zolkis, 2018-09-29:
> Does creating it have to be async? Your last IDL block has it as promise-returning, and it's not clear to me why.

The [watch()](https://w3c.github.io/web-nfc/#dom-nfc-watch) algorithm was asynchronous, but indeed the user intent to create a read filter does not need to wait until NFC is configured, so we could create the object right away. We could handle security/permissions/support errors with exceptions. We could eventually include an `onerror` event on the `NFCReadFilter` object to report NFC/read errors.

zolkis, 2018-12-04:
The name `NFCReader` confused me a bit, since it's meant to be only a read _filter_ originally.

kenchris, 2018-12-04:
Sure, in hardware, but you can create as many instances as you want, same with the writer. I don't have better suggestions. A file reader is also only reading one file :-)

reillyeon, 2018-12-05:
My interpretation of the name is that when the author wants to read from NFC tags they create an `NFCReader`.

There's a similar pattern in other APIs, for example, you can create multiple `Accelerometer` objects even though the hardware only has one accelerometer. The browser is responsible for mapping the web exposed object to the right platform object.

littledan, 2018-12-05:
From the outside, NFCReader and NFCWriter sound like good names to me, and the EventTarget "filter" pattern makes sense. I agree with @reillyeon that instances of a class don't need to correspond 1:1 with pieces of hardware.

Thinking of a filter that outputs an EventTarget reminds me of patterns with Observables, which might be nice here from a perspective making an interface which works in Node.js as well, though WebNFC is a bit too tied to origins to make sense unchanged there. (I'm happy to see that the origin model is working out here!)

zolkis, 2018-12-05:
I like the Observables pattern and `NFCTagObserver` name as it's clear and the semantics is conveyed more precisely, but if there is no confusion about the  `NFCReader` name (meaning actually read-filters), `NFCReader` seems to be more aligned with the Web Platform.

closed 2018-12-13 by kenchris