w3c-cg/web-nfc #156: Do not use "script execution environment"

domenic, 2018-09-29:
"script execution environment" is not a well-defined term. The spec links to https://html.spec.whatwg.org/#script-execution-environment which does not exist. It then seems to try to define it itself, but in a way that doesn't actually say what it is.

I'm not sure what the point of this term is, but it seems to be only used in the non-normative security policies section, plus once redundantly elsehwere ("Each Window object connected to the script execution environment" can just be "Each Window object") so hopefully it can be removed.

littledan, 2018-12-05:
The NFC global state has to be associated with something or other. Would it make sense to hang it off the environment settings object?

kenchris, 2018-12-05:
Yes, we are moving towards using constructors and getting rid of the NFC interface, so we need somewhere to store the internal slots

PR #178 (2018-12-06): Editorial: Move state from the NFC interface to the settings object
This refactoring should make it easier to change the interface to
NFCReader, to address #152

Along the way, closes #156.
commit aaad6d12 (2018-12-06): Editorial: Move state from the NFC interface to the settings object

This refactoring should make it easier to change the interface to
NFCReader, to address #152

Along the way, closes #156.
closed 2018-12-06 by kenchris