w3c-cg/web-nfc #159: Web-NFC status question

tahv0, 2018-10-04:
Hi,

What is status of web-nfc regarding support on different devices and browsers? I tried on my Android device and it seems to work fine, but will this come to iOS devices any time soon? And does this work on commonly on any Android device?

Links I found myself
https://www.chromestatus.com/feature/6261030015467520
https://stackoverflow.com/questions/51134691/can-the-web-nfc-api-be-used-in-progressive-web-applications

Thanks.

cyberphone, 2018-10-04:
I haven't tried Web NFC yet but it seems that in order to use it you must first "surf" to the proper application.  This differs from native NFC applications which are invoked automatically.

anssiko, 2018-10-04:
Thanks for testing the Web NFC API!

Regarding iOS support, currently WebKit Feature Status https://webkit.org/status/ does not list the Web NFC API, but there's a feature/bug report open at http://webkit.org/b/163196

(For example, Microsoft's Edge Platform Status https://developer.microsoft.com/en-us/microsoft-edge/platform/status/webnfcapi/ lists the feature as "Under Consideration".)

Requests from web developers will help browser vendors prioritize the feature. See the bottom of the WebKit Feature Status page for information on how to inform the WebKit project this feature would be wanted.

Any Android device with NFC hardware should support the Web NFC API. The Web NFC API is implemented on top of the android.nfc API https://developer.android.com/reference/android/nfc/package-summary introduced in Android 2.3.

Hope this helps. I'll close this issue. If you have any questions or contributions related to the specification itself, please feel free to open a new issue.

closed 2018-10-04 by anssiko

tahv0, 2018-10-06:
@anssiko It feels like this web-nfc doesn't support all kind of NFCs. I tried to read some credit cards with native applications and those works fine.
Some cards with NFC returns `{"records": ["data" null, "mediaType" "", "recodType": "empty"]}`, some tags are read just fine and big part is not detected at all.
Is there fool-proof list somewhere which tells which NFCs are supported?

anssiko, 2018-10-08:
(Seems to be a duplicate of https://github.com/w3c/web-nfc/issues/145#issuecomment-427291378)

Stryker93, 2019-03-03:
Same question here: I tried severall RFID cards, like Mifare Desfire, Mirfare Ultralight C, Mifare Classic and all are just giving a null, when reading with a Samsung Galaxy Tab Active.
So is there a list which NFC cards are supported by the Web NFC API?

I simply like to read the ID of RFIDs cards in my web app and I would like to know which RFID cards I will need to buy.

BTW: A native app on the Galaxy Tab Active shows the ID of all these cards. So it does not seem to be a hardware problem.

> @anssiko It feels like this web-nfc doesn't support all kind of NFCs. I tried to read some credit cards with native applications and those works fine.
> Some cards with NFC returns `{"records": [...]

kenchris, 2019-03-03:
Currently all cards are actually follow the NFC spec, like Type 1-5. Mifare Classic for instance isn't actually a NFC tag though it supports NDEF.

We have changed the spec to support these NDEF compatible cards supported by the reading hardware, but that hasn't been implemented yet

kenchris, 2019-03-03:
Btw, the Chrome implementation isn't 100% following the spec, try adding accept: "any" as part of the options

Stryker93, 2019-03-04:
@kenchris 
I already used the option "any" :(

So we know that Mifare Cards do not work so far. But what cards DO work? The description "NFC Forum Type 1 - 5" is very theoretical: I did not find any onlineshop selling RFID cards with such descriptions. All I found had fancy names like NTAG, EM4200, Mifare, ICode,...

anssiko, 2019-03-04:
I'm wondering if it'd be helpful to create an unofficial "what works" compatibility table in this repo the community could help maintain ([example](https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md#gatt-communication-api)).

It would be an overwhelming task for the current spec editors to acquire and test all the cards in the world across all the devices.

kenchris, 2019-03-04:
All, that I have tried actually worked so far. You can install NFC tools on Android and it will tell you what type it is. So does a bit of Googling.

For instance is say here that NTAG213 215 and 216 are Type 2

https://www.nxp.com/products/identification-and-security/smart-label-and-tag-ics/ntag/ntag-for-tags-labels/ntag-213-215-216-nfc-forum-type-2-tag-compliant-ic-with-144-504-888-bytes-user-memory:NTAG213_215_216

kenchris, 2019-03-04:
Also try publishing your code samples if possible

Stryker93, 2019-03-04:
So you mean NTAG215 works with the web NFC API?

I got a sample card from a seller which is sold as NTAG 215, but a) the web NFC says "null" and b) a native app reads the ID but says it is "Mayfair Ultralight C". Hopefully the seller just shipped the wrong card and I only need to buy a "real" NTAG215 if it works with the web NFC API.

> For instance is say here that NTAG213 215 and 216 are Type 2

kenchris, 2019-03-04:
Mifare is a NXP brand. Mifare Classic are not following the NFC spec fully, but others are. If you have Android can you try the NFC app so see what type it is. It should just work with Type 1-4 (5 might not be supported yet as it is quite new and might require a newer NFC chip in your phone).

If it isn't working something else is wrong, like are you using HTTPS etc, how does your code look?

kenchris, 2019-03-04:
Let me show you a simple test that I just did with Mifare Ultralight (Type 2, uses NFC-A for communication)

[image]

[image]

[image]

Stryker93, 2019-03-04:
Code comes from https://whatwebcando.today/nfc.html

Result (for example from Mifare Desfire card):
"NFC message received from URL undefined"
Raw Message: {"records": ["data" null, "mediaType" "", "recodType": "empty"]}

Result in a native app (on the same Samsung Galaxy Tab Active):
[image]

kenchris, 2019-03-04:
It says NDEF formattable, so maybe you need to format it using that app first

Stryker93, 2019-03-04:
I only want to read the UID/serial number of the RFID cards ("Seriennummer" in the screenshot). Is there no way I can get it via web NFC API?

kenchris, 2019-03-05:
OK filed https://github.com/w3c/web-nfc/issues/212 as this is currently not available

Stryker93, 2019-03-06:
Great, thank you for adding!

Stryker93, 2019-03-09:
@kenchris Maybe a stupid question, but I am not that experienced with this whole w3c, github und release thing: any idea when the reading of the serial will be available in Android Chrome?

richardszalay, 2019-10-04:
@Stryker93 The latest Chrome supports accessing the serial via the `NFCReader` interface.