w3c-cg/web-nfc #160: Create error message when using proprietary tags

kenchris, 2018-10-10:
I am also starting to think that we might go ahead and just allow these tags with some opt-in like "proprietary" and turn the mode into maybe a bit field.

closed 2019-02-15 by kenchris

reillyeon, 2019-05-28:
@kenchris, can you provide background on why this needs to be opt-in?

zolkis, 2019-05-29:
Also related to #226. I wonder could we make NDEFCompatibility simpler (or even more, get rid of it).
I would not care about the reader type once the content is NDEF. Could we leave compatibility issues to implementations, or do the applications/web pages need to have the option to make a choice there? (even then, a sensible default would be to "not care, just read/write the content if you can").

kenchris, 2019-05-29:
@reillyeon the background is that there are NFC-like tags that offer NDEF etc but don't work with all readers as they are not following the standard. Mifare Classic is a common example of this and it can only be written to and read by phones that have an NFC chip made by NXP. If it for instance has a NFC chip made by say Sony (like my old phone) they these tags cannot be read at all.

I think that comes as a surprise to many people as these tags are often sold as just being NFC tags, though they are not, so this sets the expectations straight.

The tags are basically proprietary and the web is about standards so you could ague that we should ignore them. But they are quite popular and from talking to potential users of Web NFC, there are valid use-cases. For instance, I talked to a company where the keycards are Mifare Classic and they want to use Web NFC to easily tap a keycard and a NFC sticker on a room to reserve meeting rooms.

zolkis, 2019-05-29:
@kenchris so the rationale for NDEFCompatibility is that the page can tell it only wants standard tags or it wants to implementation to consider also non-standard tags if the implementation/underlying platform can deal with it.

However, why wouldn't a page always opt for the widest compatibility available from the platform? Do we have a use case when a page explicitly does not want to read/write non-standard tags? Otherwise I suggest we take the "widest possible" compatibility and for the time being remove NDEFCompatibility and explain the handling of various compatibility scenarios in prose.

kenchris, 2019-05-29:
By default you are not able to read / write to all tags and we should throw an error when they are attempting so that the devs become quite aware that it is non standard. We already got such feedback from the existing impl in Chrome, so it definitely works.

This makes the opt in very explicitly.

I also know of people who have bought Mifare Classic tags and believed they would work everywhere as it worked for them, only to later find out that is now the case

zolkis, 2019-05-29:
I am fine with keeping it explicit, especially if it has been considered and feedbacked earlier - but I wanted to understand.
OK, so when tag comms does not work, we throw an error, so the devs would know it's non-standard.
But this could be done even without NDEFCompatibility.
What I am trying to understand is whether are you trying to minimize the chance of error reporting by making the devs aware of the compatibility issues and by pushing them to make an explicit choice there? (whereas my thinking was devs would not really care until an error is received, but even then there is not much to do about it than just ack the error, since the platform can't handle it anyway).

kenchris, 2019-05-29:
Yes, we should throw such an error with good explaination and make it work (not throw error) if NDEFCompatibitity is set to `any` or `vendor`

reillyeon, 2019-05-29:
The developer education re: potentially incompatible cards argument sounds good to me. Can we add spec text clarifying that as the purpose?

leonhsl, 2019-06-04:
Is there any special concern for us to have the `vendor` compatibility option?
From reading the spec, the only reason seems to be that a reader with a `vendor` option is not allowed to read `nfc-forum` messages: step 1.5 of https://w3c.github.io/web-nfc/#dispatching-nfc-content
Is this use case a required/valid one? If not, `nfc-forum` and `compatible` seems enough for us and easy to understand for devs.

PR #189 (2019-02-15): Support opting into supporting reading and writing to all NDEF devices
The most common feedback we have gotten is the lack of support for
the super popular Mifare Classic tags. These are NDEF compatible but
not part of the NFC spec and only supported by NXP reader chips.

This adds an explicit opt-in to reading and writing to these tags

Fixes #160
commit 9dd6cb53 (2019-02-15): Support opting into supporting reading and writing to all NDEF devices

The most common feedback we have gotten is the lack of support for
the super popular Mifare Classic tags. These are NDEF compatible but
not part of the NFC spec and only supported by NXP reader chips.

This adds an explicit opt-in to reading and writing to these tags

Fixes #160