w3c-cg/web-nfc #161: navigator.nfc undefined

jdiegosierra, 2018-10-12:
When I test my phone in this web https://whatwebcando.today/nfc.html I'ts okey.
However when I use my own web navigator.nfc is undefined. 

`<!DOCTYPE html>
<html>
<head>

<title>NFC</title>

</head>
<body>

<button class="btn btn-lg btn-default" onclick="readWriteNfc()">Test NFC Read/Write</button>
<p id="log"></p>

<script>
        document.getElementById("log").innerHTML = navigator.nfc;
        function readWriteNfc() {   
            try {
                navigator.nfc.watch((message) => {
                document.getElementById("log").innerHTML = message;
                });  
             }
             catch(err) {
                document.getElementById("log").innerHTML = err.message;
            } 
        }
        </script>

</body>
</html>`

rohmanhm, 2018-10-16:
What is your phone? And what browser do you use?

anssiko, 2018-11-01:
As with most powerful capabilities, only content served over https is allowed to access Web NFC API. If you are using a local development environment, you may be running into this issue.

For the record, this is specified in:
https://w3c.github.io/web-nfc/#extensions-to-the-navigator-interface
https://w3c.github.io/web-nfc/#the-nfc-interface

`[SecureContext]` is to say the API is exposed only within a secure context that is spec-speak for https.

Also noted in the security policies section, see the first bullet at:
https://w3c.github.io/web-nfc/#security-policies

Hope this helps, and good luck with your further Web NFC experiments! I'm closing this issue. Any feature requests or spec bugs, please open a new issue.

closed 2018-11-01 by anssiko

anujwebkul, 2020-01-17:
@anssiko 
I am using chrome version 79.. but still navigator.nfc undefined and I have enabled the webnfc flag as well.

beaufortfrancois, 2020-01-17:
@anujwebkul `navigator.nfc` is not a thing anymore in Web NFC. See https://w3c.github.io/web-nfc/#examples for many examples on how to play with Web NFC. Hopefully you'll have fun!

anujwebkul, 2020-01-17:
@beaufortfrancois but when I am using reader.scan(); then it gives the below error.
DOMException: WebNFC feature is unavailable.

beaufortfrancois, 2020-01-17:
According to https://cs.chromium.org/chromium/src/third_party/blink/renderer/modules/nfc/nfc_proxy.cc?type=cs&sq=package:chromium&g=0&l=201, it looks like NFC support is not available on your device. Which Android device do you have?

riju, 2020-01-17:
@anujwebkul : Can you try on [Chrome canary](https://play.google.com/store/apps/details?id=com.chrome.canary&hl=en) and let us know if the issue still persists ?

anujwebkul, 2020-01-17:
It's working on [chrome beta](https://play.google.com/store/apps/details?id=com.chrome.beta&hl=en) like a charm, Thanks :)

beaufortfrancois, 2020-01-17:
But not Chrome canary?

anujwebkul, 2020-01-17:
@beaufortfrancois No.

leonhsl, 2020-01-19:
I tried the canary `81.0.4031.0` and `reader.scan()` works well there.

Is there a chance that you forgot to turn on the flag `WebNFC` via `chrome://flags`? @anujwebkul