w3c-cg/web-nfc #166: Consider dropping peer for MVP

kenchris, 2018-11-30:
Hard to implement on most platforms as this really seems to be handled at platform level.

It also doesn't seem to important as writing and reading tags.

But if we remove, we need to make sure that it can be added back in, so be careful when removing enums

mrj, 2018-11-30:
My application is for a phone that's running a WebNFC-enabled browser to tap another phone to pass it a URL to which to browse.

When I joined the development group for WebNFC more than three years ago, I expected peer-to-peer to be its main focus, since it's used so much for payments, and passive tags a secondary implementation. I've been astonished that it's been the other way around, and that implementation of peer-to-peer has been delayed, and now possibly dropped. I've had to patch in QR Code support in my app to tide me over.

The alternative of the Web Share API triggering Android Beam is inadequate:

1. It only shares the URL of the invoking page. The URL set in the share call is ignored, so arbitrary URLs can't be shared,
2. It requires the user make the extra step of choosing NFC delivery,
3. The share must be triggered by a user action, and
4. The share must be from an SSL page, which is annoying for development.

Is there an alternative?

Is it really that difficult to implement peer-to-peer WebNFC on Chrome on Android, or are there security concerns, or just a belief that there's no demand? I know an implementation has been substantially written, and that the Chrome team planned to fully implement this "after we ship Generic Sensor APIs", which I believe happened in March or April. Has this now been abandoned?
 
Here is the original WebNFC Issue discussing peer-to-peer support: #139
 
And here is the Chromium bug for WebNFC issues, where I've asked about peer-to-peer: https://bugs.chromium.org/p/chromium/issues/detail?id=520391

kenchris, 2018-12-03:
Payments solutions are not using peer to peer, but "card emulation" which is different. It allows a device to emulate being a tag and additionally gains access to the "secure element" on the device.

kenchris, 2018-12-03:
Maybe we need the input from @slightlyoff @thomasthedane

mrj, 2018-12-03:
Ken, whatever the implementation detail, tap payments are still a device talking to a device. I would actually prefer WebNFC peer-to-peer to be implemented as tag emulation so it would trigger URL browsing just as smoothly.

Could you explain further what makes Android implementation hard. Does it require new functionality in Android so that an app can activate the NFC chip? Can Chromium implement WebNFC peer-to-peer by doing what payment apps do and emulate a tag, just without the secure element?

I'm keen to get some form of peer-to-peer working soon, otherwise it'll likely be a wait of several more years if it's dropped from the shipped MVP.

mrj, 2018-12-03:
I've now implemented a way to share my URLs by NFC without use of WebNFC, so I'm no longer as insistent about early peer-to-peer support.

Because the URLs being shared are within my website, and because a peer-to-peer tap when Chrome is being used will activate Android Beam to send the URL of the current page, I can set a cookie on the sending browser and then open the URL to be shared in a new window. The cookie allows my website to distinguish sender from receiver and either show tap instructions or process the share.

I still don't understand why it's hard for WebNFC to use Android Beam in the same way, just with an arbitrary URL rather than the URL of the current page.

littledan, 2018-12-05:
Are there any peer use cases in the [use cases doc](https://w3c.github.io/web-nfc/use-cases.html)? It seemed to me like those were reader/writer, but I don't know this area well.

zolkis, 2018-12-05:
We have one [here](https://w3c.github.io/web-nfc/use-cases.html#sending-image-to-another-web-nfc-capable-device) but not a very relevant one. In general, I would like to support the peer to peer use cases (like @mrj mentioned) if possible & feasible & the underlying platform exposes the interaction.

kenchris, 2019-02-19:
Let's drop this for now

closed 2019-02-19 by kenchris