w3c-cg/web-nfc #197: Help with wording

kenchris, 2019-02-20:
Hi @annevk @domenic and @littledan 

Maybe you can help me a bit with the proper wording for the following, as I am trying to tighten up our algorithms and wordings - thanks! Any help appreciated

- Let result be the result of invoking ASCII decode on the consecutive offset number of bytes.
- Set record.data to a new ArrayBuffer object constructed from the bytes of ndef.PAYLOAD.
- Set record.data to a new ArrayBuffer object representing the ndef.PAYLOAD byte sequence.
- Set the ndef.TYPE field to "T" (value 0x54 following NFC binary encoding).
- Set the consecutive bytes of ndef.PAYLOAD to the value of language encoded as US-ASCII.
- Set bit 6 to 0. (the the value 0?)

@zolkis I don't understand what you mean with this
- Set bits 5 to 0 to the length of the language string.

annevk, 2019-02-20:
I tried searching for "Let result be" in https://w3c.github.io/web-nfc/ but it doesn't yield anything. Context would help.

kenchris, 2019-02-20:
Let MIME type.parameters["lang"] be the result of invoking ASCII decode on the consecutive offset number of bytes.

annevk, 2019-02-21:
So I don't understand what "consecutive _offset_ number of bytes" means. And _offset_ is initialized in a later step, meaning this would result in some kind of specification-exception?

kenchris, 2019-02-21:
Yes, I noticed that too and I am fixing this in https://github.com/w3c/web-nfc/pull/198

kenchris, 2019-02-22:
These have been reworded now after comments in PRs

closed 2019-02-22 by kenchris