w3c-cg/web-nfc #200: There's no such thing as current browsing context. A browsing context also has no URL path.

kenchris, 2019-02-21:
There's no such thing as current browsing context. A browsing context also has no URL path.

kenchris, 2019-02-21:
Comment from @annevk:

> So you need to figure out what object you want the URL from and find a way from where you are now to there. I don't really see how browsing contexts relate here since they don't carry URLs.

kenchris, 2019-02-21:
So we have the `environment settings object` which has a https://html.spec.whatwg.org/multipage/webappapis.html#responsible-document

Which has location which has a path.

@zolkis I assume this is what we are after?

kenchris, 2019-02-21:
But then there are comments like 
[image]

So I guess what we need is `the environment settings object's responsible document's URL path`

annevk, 2019-02-21:
You need a particular environment settings object (relevant or current I suspect), though you could also use a particular global object. "URL path" is also nothing, strictly speaking. You need a URL's path.

commit a7e4aa9d (2019-02-22): Fix Web NFC id algorithm, fixes #200
closed 2019-02-22 by kenchris