w3c-cg/web-nfc #209: Is anyone using WebNFC successfully?

M600LT, 2019-02-28:
Is anyone using WebNFC API successfully?

I am attempting to follow the example here: https://whatwebcando.today/nfc.html

Blackview BV6000s Android phone. Chrome browser with both required flags enabled.

The code fails with "nfc" not being found in the navigator object. 

Thanks in advance for any advice.

M600LT, 2019-02-28:
Sorry, after more research I found the section on needing to serve the pages over https, hopefully this will resolve the issue.

anssiko, 2019-03-01:
That is correct, the Web NFC API is only exposed to pages served over https.

(Protip: when you see the [SecureContext](https://heycam.github.io/webidl/#SecureContext) keyword atop the interfaces in the specs, you know this restriction applies.)

closed 2019-03-01 by anssiko