w3c-cg/web-nfc #212: Add method to reading event to get serial if available

kenchris, 2019-03-05:
We can easily get this via Android and other platforms. Here is the Android info:

[image]

Stryker93, 2019-03-06:
Great, thank you for adding!

kenchris, 2019-03-06:
@domenic how do you think we should expose these? They seem to be either 4 or 7 bytes long, so we should return a sequence? or as a string with either `-` or `:` separating the numbers?

kenchris, 2019-03-06:
In USB it is a UCHAR and it is exposed in Web USB as a `DOMString`: https://wicg.github.io/webusb/#dictdef-usbdevicefilter

@reillyeon

kenchris, 2019-03-06:
I think a string is easiest to deal with from JavaScript, so went for a ASCII hex digit string separated by `:`

commit e8e89fc7 (2019-03-06): Add serialNumber, fixes #212
closed 2019-03-06 by kenchris