w3c-cg/web-nfc #215: Out of date statement about "NFC" object in section 9.3

DonnaWuDongxia, 2019-04-01:
As the "NFC" has been deleted, following words seem like be out of date in section 9.3:

"Each Window object where NFC is exposed has a separate NFC instance. The visibility and focus state of the Window object determines the suspended state of the associated NFC instance. "

anssiko, 2019-04-02:
Thanks for reporting @DonnaWuDongxia!

I believe this is a leftover from spec update that split `navigator.nfc` into global `NFCWriter` and `NFCReader` constructors.

Cc @kenchris

kenchris, 2019-05-23:
Yes, feel free to send PRs for this

kenchris, 2019-08-05:
@Honry didn't you fix this?

Honry, 2019-08-06:
Ah, I didn't realize there's already an issue for old "NFC" instance, sorry for the duplication.
Fixed in https://github.com/w3c/web-nfc/pull/273, let's close it now.

closed 2019-08-06 by kenchris