w3c-cg/web-nfc #216: beginners tutorial

stephomn, 2019-05-02:
Hi everybody,

well, I'de like to start using NFC under Chrome/android.
I enabled the NFC support in chrome flags but when I try
`const reader = new NFCReader();`
Console says Uncaught ReferenceError: NFCReader is not defined.
where is the class NFCReader and NFCWriter ?

Where can I find a running example ?

thank's

riju, 2019-05-02:
Thanks for checking. We are in the middle of implementing these recent changes @kenchris has been fleshing out. I can update the status back in a couple of months.

stephomn, 2019-05-03:
IHi sorry, I did not understand. It's projetc only, nothing running now ? Some one says he got something running in subject #145 "how to get started". Is there a full running example ?
I'de like to get the RFID unic ID : UID.
Most of people are intersted by tags but I need the UID first of all. I have to indentify the RFID card.
Is it possible to get  UID with integrated NFC reader on android using browser and javascript ? Is there something running now or shall I wait more ?
thank's

zolkis, 2019-05-03:
>  enabled the NFC support in chrome flags but when I try
const reader = new NFCReader();

This is the new API and has not been implemented yet.
The previous API is specified [here](https://cdn.staticaly.com/gh/w3c/web-nfc/336e6afe18470caeca47967413acb39a34c86fcd/index.html) - that should work under the experimental flag in Chrome Android.

stephomn, 2019-05-06:
Ok thank's I understand now. I'm so sorry that there is no way to get the UID of the RFID card. It is very important to identify the card to know what to do with it. Without UID an essential information is missing according to me.

anssiko, 2019-05-08:
I believe the [`NDEFMessage.serialNumber`](https://w3c.github.io/web-nfc/#dom-ndefmessage-serialnumber) property in the latest spec should satisfy this requirement. Chrome does not yet implement the latest spec, but an earlier version that lacks this property.

We're in process of implementing the latest spec. You can watch https://crbug.com/520391 for implementation progress.

I'm closing this issue. Please reopen and provide more context if I misinterpreted the request.

closed 2019-05-08 by anssiko

dizhu1991, 2019-07-08:
@zolkis The previous API is not actually working in the web app I am looking at. The data is not being read into the `navigator.nfc.watch` part, so there is no output.

The new API is not working either, as I got the same `NFCReader undefined` error after updating the code.

riju, 2019-07-08:
> @zolkis The previous API is not actually working in the web app I am looking at. The data is not being read into the `navigator.nfc.watch` part, so there is no output.
> 
> The new API is not working either, as I got the same `NFCReader undefined` error after updating the code.

@dizhu1991 Thanks for checking. We are in the midst of implementing the "new" API in chromium. Implementation status can be tracked in http://crbug.com/520391.

boatbodger, 2019-08-29:
I would just like to add my two-pennyworth that having the tag UID is essential for my use case in which I'm using a nodejs application to register tags with a database.
I have managed to get the new API to work with Chrome 78 (Canary), but there seem to be several areas where there is 'work in progress' as at today's date.
(I have managed to get the UID from a tag via a native Android application, but I don't really want to write my whole app as native)