w3c-cg/web-nfc #217: Typos in chapter 9 of the spec

leonhsl, 2019-05-05:
http://w3c.github.io/web-nfc/#the-nfcreader-and-nfcwriter-objects

"The NFCReadingEvent is the event being dispatch on new NFC readings, "
                                                                          |
                                                                   dispatched?

"The NFCErrorEvent is the event being dispatch on errors, "
                                                                     |
                                                             dispatched?

leonhsl, 2019-05-05:
Step 9.12 in this section http://w3c.github.io/web-nfc/#the-push-method

Duplicate "is":
> if device is is not officially supported by the NFC Forum, compatibility is "vendor" or "any".

leonhsl, 2019-05-05:
http://w3c.github.io/web-nfc/#listening-for-content

Duplicate "attach an"?
>  the client needs to attach an attach an event listener for the "reading" event

leonhsl, 2019-05-05:
http://w3c.github.io/web-nfc/#steps-receiving

"step 7 of the NFC listen algorithm" does not exist, instead, I guess we want to refer to step 3.2 and 3.5 of the NFC listen algorithm.
> If there are any NFCReader instances in activated reader objects then UAs MUST listen to NDEF messages, according to step 7 of the NFC listen algorithm.

kenchris, 2019-05-05:
Feel free to submit a PR for these

closed 2019-05-06 by leonhsl