w3c-cg/web-nfc #219: NFCReader should contain those members of NFCReaderOptions?

leonhsl, 2019-05-05:
http://w3c.github.io/web-nfc/#constructing-an-nfcreader-object

> For each key → value of options:
If key equals "url", set reader.[[Url]] to value.
Otherwise, if key equals "recordType", set reader.[[RecordType]] to value.
Otherwise, if key equals "mediaType", set reader.[[MediaType]] to value.
Otherwise, if key equals "compatibility", set reader.[[Compatibility]] to value.

However, NFCReader definition does not have such members: Url, RecordType, MediaType, and Compatibility.

kenchris, 2019-05-05:
These should be internal slots on NFCReader yes

leonhsl, 2019-09-05:
Now the new place for this problem is at http://w3c.github.io/web-nfc/#the-scan-method.

Honry, 2019-09-05:
@leonhsl, @kenchris, I am working on it. :)

PR #345 (2019-09-05): Define internal slots for NFCReader object
and make internal slots linkable

Fixed #219

***
<a href="https://pr-preview.s3.amazonaws.com/Honry/web-nfc/pull/345.html" title="Last updated on Sep 5, 2019, 7:11 AM UTC (f6dad9e)">Preview</a> | <a href="https://pr-preview.s3.amazonaws.com/w3c/web-nfc/345/a33dbb1...Honry:f6dad9e.html" title="Last updated on Sep 5, 2019, 7:11 AM UTC (f6dad9e)">Diff</a>
commit 1c4ac8cc (2019-09-05): Define internal slots for NFCReader object

and make internal slots linkable

Fixed #219
closed 2019-09-05 by kenchris